prusa3d/PrusaSlicer #1355: [MMU2] Wrong purge at end of print

acorderob, 2018-10-27:
### Version
1.41.0+win64

### Operating system type + version
Windows 10

### Behavior
When printing a multimaterial object whose ending layers are only one material, at the end of the print there is a purge in the wipe tower, but the tower stopped growing in the last multimaterial layer, so it purges in mid air. I expect it to not do a purge at the end of the print since it doesn't seem necessary.

[image]

#### STL/Config (.ZIP) where problem occurs
[issue.zip](https://github.com/prusa3d/Slic3r/files/2521654/issue.zip)

bubnikv, 2018-10-27:
That is the final ramming before the filament is pulled from the nozzle and parked into the filament switcher. This is a FAQ. Closing.

closed 2018-10-27 by bubnikv

akust, 2019-04-07:
same issue here but I don't use prusa MM but Palette2, no "ramming" and "parking" needed, it plans to spit out filament into air - threatening the print, is there an option to avoid that?

lukasmatena, 2019-04-08:
@akust Ramming can be turned off in filament settings (Filament Settings -> Advanced->Ramming settings... and then shorten ramming to zero). It of course turns off the ramming for the whole wipe tower, which is desirable in your case, as far as I understand it.

The parking position of the filament and similar stuff is configurable in Printer Settings -> Single Extruder MM Settings (the group is only shown when Single Extruder MM  is set on General tab.

Let us know if it's flexible enough for your needs.

akust, 2019-04-13:
great, thank you for the detailed informations!

sidekickc, 2022-06-24:
> That is the final ramming before the filament is pulled from the nozzle and parked into the filament switcher. This is a FAQ. Closing.

There is no FAQ called Closing.

sidekickc, 2022-06-25:
> https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)

Thanks. I suspect the FAQ called "Why there is an extrusion hanging in the air over a multi material wipe tower?" used to be called "Closing", so this would be causing some confusion, as it did for me.

dwprod38, 2025-02-20:
> [@akust](https://github.com/akust) Ramming can be turned off in filament settings (Filament Settings -> Advanced->Ramming settings... and then shorten ramming to zero). It of course turns off the [...]

I've yet unchecked "Enable ramming for mutlitool setups" --> Nothing changed...

[image]

To be investigate...