prusa3d/PrusaSlicer #1356: Slicing produces levitating models

pentacular, 2018-10-27:
### Version
Slic3rPE-1.41.1+linux64-full-201810261150

### Operating system type + version
Linux 4.17.0-3rodete2-amd64

### Behavior
* Produces empty bottom layer, causing printer to try to print levitating model.
* Reproduce by import model, use mk3 / 0.20mm fast, slice, look at bottom layer.
* Expect bottom layer to be removed rather than empty, so that model is not levitating.
* Bottom layer is empty
[image]

#### STL/Config (.ZIP) where problem occurs
[bug.zip](https://github.com/prusa3d/Slic3r/files/2521772/bug.zip)

jmd73, 2018-10-27:
Just to report, with Windows 10 64bit and Slic3r 1.41.1+win64 final version, it slices as it should with the same settings.

[image]

pentacular, 2018-10-27:
Please check the bottom-most layer.

jmd73, 2018-10-27:
Oh! Now I see what you mean. I was wrong.

If you activate "Detect thin layers" you get the bottom too, but it's only one thin line to build the rest of the object on. I think the object should have a "flat" are on the bottom, or be printed with the wide opening down.

pentacular, 2018-10-27:
I know how to work around it, but if slicing is intended to produce an
object resting on the bed it should do so.

In this case it decided the bottom layer was empty and to rest the object
on that empty layer.

It should have decided to try to rest the object  on the lowest non-empty
layer instead.

rtyr, 2020-03-29:
An extra check was added since PrusaSlicer 2.1.0, so unprintable scenarios like this are detected. Closing.

[image]

closed 2020-03-29 by rtyr