prusa3d/PrusaSlicer #1357: Deleteing slic3r.ini %user%\AppData\Roaming\Slic3rPE

drummonds, 2018-10-27:
### Version
_Version of Slic3r used goes here_
1.41.0 win64

### Operating system type + version
_What OS are you using, and state any version #1325 
Windows 10.0.17134 64 bit 

### Behavior
Slic3r wouldn't start.
Not only me who has the problem: https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/slic3r-pe-1-40-won-t-open-t19726.html

slic3r.ini was empty.
I replaced with an old slic3r.ini that I had and a little configuration and working again.

Ideas:
- Create a backup version
- Try backup version if main is corrupt
- Recognise problem and don't just hang.

(I do like it - using with an old ormerod)

vojtechkral, 2018-11-08:
Indeed, and a similar issue happens when `slic3r.ini` contains garbled data, which for some reason sometimes happens on Win10. We'll fix this.

bubnikv, 2019-08-24:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/2438

While we have no explanation on how the PrusaSlicer.ini file could have gotten corrupted, we suspect that the Windows API for renaming files is not atomic and it may in some cases lead to file corruption. We replaced the "file rename" function with a complex function from the llvm project with fd3fe75. Let's hope it will fix this mysterous issue.
Closing, the fix will be part of the PrusaSlicer 2.1.0-beta.

closed 2019-08-24 by bubnikv