prusa3d/PrusaSlicer #1358: Display problem

RussWues, 2018-10-27:
### Version
_Version of Slic3r used goes here_1.41.0+win

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_ 
View bar at bottom displays only 1st block (always) and the others I have seen only once. When displaying all blocks, everything in that bar is active, but not active otherwise.

* _Steps needed to reproduce the problem_
Slice object and enter preview mode.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
All blocks on bottom view bar are active and working.

* _Actual Results_
Only 1st block highlighted and selections can be made, but no action accomplished
.
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_
Yes

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

jmd73, 2018-10-27:
You need to export G-code first, then you can see all features.

lukasmatena, 2018-10-27:
I'm closing the issue. As mentioned above, you have to export G-code before the extra controls become active. We are planning to let Slic3r export gcode in the background to eliminate this annoyance. We have already promised this quite a long time ago, now it's already working in our development branch and it will be part of the next release (1.42).

closed 2018-10-27 by lukasmatena