prusa3d/PrusaSlicer #1359: feature request: show print time estimation while slicing and not on gcode export

photog0411, 2018-10-28:
I rely on the print time feature a lot to determine what settings I am going to use. But it's kind of inconvenient that it only shows it in the software when you export the gcode to the sd card. It would be nice if it could do it while you are slicing or changing the configuration. Would save a lot of time.

lukasmatena, 2018-10-28:
We are planning to let Slic3r export gcode in the background to eliminate this annoyance. We have already promised this quite a long time ago, now it's already working in our development branch and it will be part of the next release (1.42).

cnlson, 2018-10-28:
> We are planning to let Slic3r export gcode in the background to eliminate this annoyance. We have already promised this quite a long time ago, now it's already working in our development branch and it will be part of the next release (1.42).

I am pretty sure @photog0411 means something different. although background export sounds nice and troublesome.

I think what was wanted is to be able to quickly see how the time is affected when changing a setting for instance, adding support, or changing infill, or shells or ... 

So is it possible to create that absent the gcode being created? 

on the background export, will I be able to hit the export button and then move things around and delete them and replace with other shapes and immediately hit export again as soon as it finishes in the background? even right now when it is exporting i sometimes miss that is not done and start working on slicing the next item only to be stopped with "another export is currently in process" which then never finishes and i have to close and restart everything. So i am really interested on whether it will allow that or not.

lukasmatena, 2018-10-29:
@cnlson It is necessary to export gcode in order to calculate that kind of information, there is no easy way past this. Even a small change can have big consequences, there are features like auto-cooling, extra toolchanges with MM printing and so on. Not to mention that the way the print estimator works internally relies on the gcode being available. It is one of the reasons it works so well. Same applies to the advanced gcode preview.

But you need not worry. The fact that reslicing during gcode export hangs the application is a bug which will be fixed. So you will be able to do whatever you want whenever you want.

RudolphRiedel, 2018-11-07:
I just came here because I have the same issue, the need to write the gcode to disk in order to check out the effect of changes to the config is rather inconveniant.

Why does the gcode has to be exported for the time estimate?
Because the gcode is used to calculate times?
Then why not generate the gcode in a buffer and use it from there and change the export function to, well exporting that buffer?

lukasmatena, 2018-11-07:
@RudolphRiedel I suggest you read all posts in an issue before you post a new one. I believe all answers are here already. If not, please clarify the question.

RudolphRiedel, 2018-11-07:
Just the same issue, take it as +1. saved me from creating a new feature request.
And I still wonder how this will work because I can not imagine how "export gcode in the background" is the answer to the issue.

photog0411, 2018-11-07:
It's been years since I used simplify3d but doesn't it tell you the estimated print time before you export the gcode file? That's what I'm looking for. I think it shows you the estimated print time when you are previewing the layers. 

I'm not a computer programmer and I'll say I don't really know how slic3r works, but if it can show you the tool paths, layer speeds, etc in the preview tab can't it give an estimate to time?

lukasmatena, 2018-11-07:
@RudolphRiedel
Now you must click the 'Export gcode' button to trigger the calculation of print time estimate etc. In future this gets triggered automatically and you won't have to click anything. That's how it solves the issue.

@photog0411 I don't know what Simplify3D does, but as repeatedly mentioned, this will be improved to get the behaviour that you describe. So yes, it of course can give you the time estimate. You're only wrong in your implicit assumption that it must be easy to do.

bubnikv, 2018-12-12:
Automatic generation of the G-code into a temp file & showing the info automatically will be part of the upcoming 1.42 alpha release. Closing.

closed 2018-12-12 by bubnikv