prusa3d/PrusaSlicer #1360: Feature request

martinowren, 2018-10-28:
### Version
1.41.0

### Operating system type + version
Windows 10

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

mloidl, 2018-10-28:
IMO this makes no sense. Since this should and can be done on the printer only.
I think you want to set the first Layer Height. This is already implemented.
[image]

eboston, 2018-10-28:
I think what is being asked for is a way to change the internal Live Adjust Z value to a different value.  While the Prusa's PINDA probe has a temperature adjustment to compensate for different bed temperatures, other printers do not.  I have seen where I have to adjust the Live Adjust Z value based on different materials.  PETG needs a different value than PLA for example.

StephenRC, 2018-10-28:
On Sun, 28 Oct 2018 06:55:50 -0700, eboston <notifications@github.com>
had a flock of green cheek conures squawk out:

> I think what is being asked for is a way to change the internal Live Adjust Z value to a different value.  While the Prusa's PINDA probe has a temperature adjustment to compensate for different bed [...]

Then adjust the z offset in the slicer start gcode.

I have a DuetWifi and use a macro to set the appropriate z offset in
different 'printer settings' for the printer & filament used.

-- 

Stephen

CONGRESS.SYS Corrupted: Re-boot Washington D.C. (Y/N)

---
This email has been checked for viruses by AVG.
https://www.avg.com

bubnikv, 2018-12-12:
As @StephenRC proposed, you may use a macro to control the Z offset based on the print & filament used. You may ask him for details of his solution, you will find the description of the macro language here:
https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language
Closing

closed 2018-12-12 by bubnikv