prusa3d/PrusaSlicer #1361: Bridges often have no contact with 0% infill due to solid infill gap

KlausHans, 2018-10-28:
### Version

1.41.1 

### Operating system type + version
Ubuntu 18.04

### Behavior
* _Describe the problem_
I want to print a model with zero infill, because the bridging distance is quite short. The model has rounded edges at the top. The Problem is, that due to the round edges Slic3r adds small solid infill areas. These areas get partially printed after the bridges are printed. Of course this means, the bridges have no contact to the shell in this case.
* _Steps needed to reproduce the problem_
Print a model with round edges at the top, use 0% infill.
* _Expected Results_
The bridges are printed after all solid infill areas are printed, so they get a nice contact.
* _Actual Results_
The bridges are sometimes printed before the solid infill additions so they have no contact on some parts.
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_
It not a feature, its a bug imho.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
I will provide it if necessary.

bubnikv, 2018-10-29:
This is a similar issue to https://github.com/prusa3d/Slic3r/issues/442

bubnikv, 2018-10-29:
Anyway, this object is simply not printable without infill. This issue is quite an evergreen. You would like Slic3r to reduce the number of inner perimeters to anchor the bridge, which certainly makes sense, but it violates the "number of perimeters" requirement you gave to slic3r elsewhere.

KlausHans, 2018-11-01:
Thank you for the reply. Yes it seems similar, but its not quite the same (see explanation to problem no 1 below).

Hm, i don't understand your explanation. In the example above i have set 3 perimeters and they are there (2 inner yellow, one outer orange perimeter). And i don't see why it should be impossible to print. It works sometimes and the bridges look good. The problem is the order of printing. The bridges need to be printed last.

Here is a small example model:
[test.zip](https://github.com/prusa3d/Slic3r/files/2539697/test.zip)
Print settings:
- 0% Infill
- 3 perimeters
- 5 top solid layers
- 5 bottom solid layers

Here is the print. In this case it causes two problems.
First:
[image]
The upper part of the bridge area has no contact, because the small inner infill was printed after the bridge. The bottom part of the bridge area (sorry for out of focus) was printed perfectly because the solid infill additions were printed beforehand. Both sides are symmetrical shaped. If it were a problem as you said, no side would have contact.

This second problem is exact the same issue as #442:
[image]
Additional screenshot of Slic3r for the second problem:
[image]
Same part and layer with 0.1% infill:
[image]

So far my data.

supermerill, 2018-11-21:
see https://github.com/prusa3d/Slic3r/issues/1185

can you test my fork to see if my fix works?

x0rtrunks, 2018-12-19:
I don't think this is the same as #1185 because I have seen it happen when infill is greater than 0%. What happens is that when you have ensure vertical shell thickness enabled it will replace a full top layer with a gap fill, but if this happens on a layer that also needs bridging infill, it will print the bridging infill before the gap fill leaving the bridging infill nothing to attach to. It's not consistent either, sometimes it will start with the gap fill and then switch to bridging, and then switch back to gap fill in order to reduce travels and optimize print time, but really if there is bridging infill on a layer it should complete all of the gap fill before moving on to the bridging so that there are proper anchors in place to connect to. As it is now, the areas that get gap fill first bridge fine, the areas that don't fail. I hope I explained that correctly. The screenshots above illustrate the problem perfectly. I believe this is a new bug within the last few releases as this wasn't always the case.

KlausHans, 2020-03-29:
Wohoo, its fixed in 2.2.!
2.1:
[image]
2.2:
[image]

closed 2020-03-29 by KlausHans