prusa3d/PrusaSlicer #1363: 1.41-beta+win64 skips in Z

Zephrant, 2018-10-28:
### Version
SlicerPE 1.41-beta+win64

### Operating system type + version
Windows 7

### Behavior
 It decided to effectively skip layer 83, moving in Z from 16.4 to 16.8 (0.2mm layers requested), which gave me a gap that caused the part to easily break at that layer. 
It did it again at 28.4-28.8 at layer 141, 41.4-41.8 (204) and so on up the part. Every time it hit a specific geometry, it skipped a layer. Note the missing layers in the pictures.

In the second picture, it shows blue for Overhang Perimeter, when it should just be solid infill.

Appears to generate code just fine in 1.40-beta-Win64 (picture 3).

#### STL/Config (.ZIP) where problem occurs

[image]
[image]
[image]

bubnikv, 2018-10-28:
please retest with the 1.41.1 final release.

Zephrant, 2018-10-28:
Unable to duplicate in 1.41.1+win64. Will repost if it shows up again.

closed 2018-10-28 by Zephrant