prusa3d/PrusaSlicer #1366: 1.41.0 + .1 remaining times export failed. cannot open file for reading  Export failure

cnlson, 2018-10-28:
### Version
_Version of Slic3r used goes here_
1.41.0 and after upgrade .1

### Operating system type + version
_What OS are you using, and state any version #s_
win7 pro x64

### Behavior
* _Describe the problem_ when set to include remaining times longer prints fail when exporting 
* _Steps needed to reproduce the problem_ load a large stl and slice, then attempt to export g-code
* _Expected Results_ should slice
* _Actual Results_ eror message, giant temp file (50m on up) 
*_Workaround Turn off remaining time in printer tab and slice
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_ bug fix

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[Giant_Posable_Halloween_Spider_w_REMIXED_leg_segment_UPDATED.zip](https://github.com/prusa3d/Slic3r/files/2523217/Giant_Posable_Halloween_Spider_w_REMIXED_leg_segment_UPDATED.zip)

bubnikv, 2018-10-29:
Most likely you have some file monitor in action (file explorer with its previews, some anti-virus or whatever else). 

GCodeTimeEstimator::post_process_remaining_times() opens the temporary G-code file with 
 boost::nowide::ifstream in(filename);
At that time, most likely the file monitor already opened the file for reading. We should open the file for reading as well, my bet is, that we do open the file for reading & writing.

bubnikv, 2018-10-29:
These are the Win32 CreateFile attributes used by Slic3r at the point where the CreateFile call must fail on yur computer. 

[image]

Looking at the dwShareMode parameter
https://docs.microsoft.com/en-us/windows/desktop/api/fileapi/nf-fileapi-createfilea
it seems like Slic3r is not guilty your problem, but some file monitor application that you installed, most likely an anti-virus. Would you please confirm it?

bubnikv, 2018-12-12:
No response, closing.

closed 2018-12-12 by bubnikv