prusa3d/PrusaSlicer #1367: Associate gcode files to stl files

duartemv, 2018-10-28:
### Version 
1.41.1+win64

### Operating system 
Windows 10 Professional

### Behavior
When slicing an STL file, if I close and reopen Sllic3r, it does not recognize that there is a gcode file generated by that STL.

It would be a great feature if the gcode was associated to the STL file that was used to create it.
I have this feature in ReplicatorG, which a much older software, and I use it regularly for an older printer. It is very useful because I can see the details of the generated gcode when I open the STL file, and make the necessary changes.

_Is this a new feature request? Yes

Thanks,
Duarte

supermerill, 2018-10-29:
And if you create many gcode from one stl?

I don't understand how it works and what are the advantage, can you write a detailed use-case?

supermerill, 2018-10-29:
You can do "Plater"->"export plate as 3mf" each time you export a gcode file.
The 3mf format should keep your settings & stl file for easy re-playability.

duartemv, 2018-10-30:
Didn't know that. I'm used to using STL files only.

I'll try that method.

Thanks for your support and keep up the good work!