prusa3d/PrusaSlicer #1369: Uneven layers on specific area

alientek, 2018-10-29:
### Version
1.41.0

### Operating system type + version
Win 10

### Behavior
Smushed / uneven layers as soon as there is a big flat surface.

[image]

supermerill, 2018-10-29:
you can try to reduce the infill/perimeter overlap %.

I will be happy if you can test a new parameter i created on my [fork](https://github.com/supermerill/Slic3r/releases) : it's "enforece 100% fill volume" on layers->horizontalshells". This should prevent over-extrusion and under-extrusion of solid infill due to the pattern (mostly rectilinear) and randomness. If you can test it and report back if it works or not, it will be a very useful information for me.

You can also reduce the extrusion multiplier, maybe your filament id just a bit too thick, and it shows it on solid layers.

ManuelMcLure, 2018-12-12:
You might also want to look at your cooling fan settings. I found similar issues when printing Benchys (a "waterline") when I first switched from Cura to Slic3r and found that they were caused by the automatic cooling fan management turning off the fan completely for the layers around the Benchy's deck. Changing the fan settings to never go under 50% went a long way towards solving the problem.

closed 2026-09-29 by prusaslicer-bot[bot]