prusa3d/PrusaSlicer #1370: Flow rate

Mike01Hu, 2018-10-29:
### Version
1.41.0 + win64

### Operating system type + version
Windows 10 Home latest build

### Behavior

From reset/switch-on, the flow rate under the Tune menu item seems to be set to 95% for the selected job despite the first layer width being set to 0.45mm and the extrusion multiplier set to 1.

[Nest_Stand_1_0.2mm_PLA_MK3.zip](https://github.com/prusa3d/Slic3r/files/2524627/Nest_Stand_1_0.2mm_PLA_MK3.zip)

supermerill, 2018-10-29:
Are you sure it's a slic3r problem and not a firmware one?

slic3r: it's the software on your pc where you put a stl file and output g-code.
firmware: it's on your printer.

here is the issue tracker for slic3r
the issue tracker for firmware problem is [here](https://github.com/prusa3d/Prusa-Firmware)

bubnikv, 2018-12-12:
MK3 profiles set the flow rate to 95%, you will find the following line in your custom start G-code:
M221 S{if layer_height==0.05}100{else}95{endif}

This is to correct for the tiny gaps between the extrusions, which cannot be easily filled due to the high viscosity of the melted plastic.

Closing.

closed 2018-12-12 by bubnikv