prusa3d/PrusaSlicer #1371: Feature Request: modify size of generic modifiers

alexiri, 2018-10-29:
The dimensions of a generic modifier can only be set when the object is created. If you set incorrect dimensions, you have to delete the object, lose all your settings and create a new one. This is really annoying, it would be much easier if the dimensions could be changed after creation.

BETLOG, 2018-11-09:
+1
But in all uppercase and enormous text. Very overdue, all the cool tweak features relating to this are currently incredibly tedious to use because of the lack of standard transformation manipulators in the modifier editor.

lukasmatena, 2018-11-09:
You are both right that the current way of editing the modifier volumes is inadequate. The improvement may seem overdue, but please understand that it requires significant effort, it is not a one-line fix. That being said, it is coming in the 1.42 release, which should make it much easier to work with modifier objects. It will be possible to edit the them from the main 3D scene in essentially the same way as you edit any other object.

BETLOG, 2018-11-10:
Is there a general consultation thread/issue for the modifier editor?

bubnikv, 2018-12-12:
Manipulation of the modifier meshes is much improved in the upcoming Slic3r PE 1.42 alpha. Closing.

closed 2018-12-12 by bubnikv