prusa3d/PrusaSlicer #1372: AppImage ends with return code 1

aboettger-tuhh, 2018-10-29:
### Version
1.41.1+linux64-full-201810261150

### Operating system type + version
linux64

### Behavior
AppImage ends with return code 1

```
execve("./Slic3rPE-1.41.1+linux64-full-201810261150.AppImage", ["./Slic3rPE-1.41.1+linux64-full-2"...], 0x7fffc1627b90 /* 84 vars */) = 0
brk(NULL)                               = 0x24a4000
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=220052, ...}) = 0
mmap(NULL, 220052, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f45e8678000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0000\21\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=18656, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f45e8676000
mmap(NULL, 20752, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f45e8670000
mmap(0x7f45e8671000, 8192, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0x7f45e8671000
mmap(0x7f45e8673000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f45e8673000
mmap(0x7f45e8674000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f45e8674000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0`l\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=149696, ...}) = 0
mmap(NULL, 132288, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f45e864f000
mmap(0x7f45e8655000, 61440, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7f45e8655000
mmap(0x7f45e8664000, 24576, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x15000) = 0x7f45e8664000
mmap(0x7f45e866a000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1a000) = 0x7f45e866a000
mmap(0x7f45e866c000, 13504, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f45e866c000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\220\37\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=116960, ...}) = 0
mmap(NULL, 2212016, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f45e8432000
mprotect(0x7f45e844e000, 2093056, PROT_NONE) = 0
mmap(0x7f45e864d000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1b000) = 0x7f45e864d000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260A\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1996592, ...}) = 0
mmap(NULL, 2004992, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f45e8248000
mprotect(0x7f45e826a000, 1826816, PROT_NONE) = 0
mmap(0x7f45e826a000, 1511424, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22000) = 0x7f45e826a000
mmap(0x7f45e83db000, 311296, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x193000) = 0x7f45e83db000
mmap(0x7f45e8428000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1df000) = 0x7f45e8428000
mmap(0x7f45e842e000, 14336, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f45e842e000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f45e8246000
arch_prctl(ARCH_SET_FS, 0x7f45e8246b80) = 0
mprotect(0x7f45e8428000, 16384, PROT_READ) = 0
mprotect(0x7f45e864d000, 4096, PROT_READ) = 0
mprotect(0x7f45e866a000, 4096, PROT_READ) = 0
mprotect(0x7f45e8674000, 4096, PROT_READ) = 0
mprotect(0x7f45e86d7000, 4096, PROT_READ) = 0
munmap(0x7f45e8678000, 220052)          = 0
set_tid_address(0x7f45e8246e50)         = 19987
set_robust_list(0x7f45e8246e60, 24)     = 0
rt_sigaction(SIGRTMIN, {sa_handler=0x7f45e86556c0, sa_mask=[], sa_flags=SA_RESTORER|SA_SIGINFO, sa_restorer=0x7f45e8661dd0}, NULL, 8) = 0
rt_sigaction(SIGRT_1, {sa_handler=0x7f45e8655750, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART|SA_SIGINFO, sa_restorer=0x7f45e8661dd0}, NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
[... 419 lines cut ...]
rt_sigprocmask(SIG_BLOCK, NULL, [], 8)  = 0
pipe([4, 5])                            = 0
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, [INT CHLD], [], 8) = 0
lseek(255, -5969, SEEK_CUR)             = 2223
clone(child_stack=NULL, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7f9f0cc4f610) = 20000
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
rt_sigaction(SIGCHLD, {sa_handler=0x55be9046c700, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7f9f0cc91100}, {sa_handler=0x55be9046c700, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7f9f0cc91100}, 8) = 0
close(5)                                = 0
read(4, "", 128)                        = 0
close(4)                                = 0
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigaction(SIGINT, {sa_handler=0x55be904690c0, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f9f0cc91100}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f9f0cc91100}, 8) = 0
wait4(-1, [{WIFEXITED(s) && WEXITSTATUS(s) == 1}], 0, NULL) = 20000
rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f9f0cc91100}, {sa_handler=0x55be904690c0, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f9f0cc91100}, 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=20000, si_uid=1000, si_status=1, si_utime=0, si_stime=0} ---
wait4(-1, 0x7ffdf719d590, WNOHANG, NULL) = -1 ECHILD (No child processes)
rt_sigreturn({mask=[]})                 = 0
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
exit_group(1)                           = ?
+++ exited with 1 +++
```

bubnikv, 2018-10-29:
What Linux distribution / distro version are you running Slic3r on?

bubnikv, 2018-11-01:
@FidelCapo would you please try to run Slic3r AppImage in a virtual machine on Ubuntu 18.10? Thanks.

vojtechkral, 2018-11-01:
@aboettger-tuhh The AppImage worked for us on Ubuntu 18.10 after isntalling missing gtk2 dependency (`sudo apt-get install gtk2.0`). Was there any error message? Does the trace follow child processes?

aboettger-tuhh, 2018-11-05:
I do not use Ubuntu after many years of use. In Arch Linux it works.

closed 2018-11-05 by aboettger-tuhh