prusa3d/PrusaSlicer #1373: Erratic behavior on bed size over 1129mm (X)

GSheltonIII, 2018-10-29:
### Version
Version 1.41.0+win64

### Operating system type + version
Windows 10 64bit 1803
OS biuld 17134.345

### Behavior
* _Describe the problem_
When creating a profile for my custom FK-1 printer (1200x400x475) any time I place a model on the print bed and attempt to move it to a different location (drag) it shoots off the bed (usually in the Y direction, but not always.) You can not drag it back to the bed. You can use the arrange button to place it back in the middle. But if you attempt to move it again, same results. I am able to manipulate the model if I set the bed to 1129x400.

* _Steps needed to reproduce the problem_
Create a printer profile with a bed larger than 1129x400, place a model, and attempt to move it.

* _Expected Results_
Expectations are for it to handle that bed size and allow me to position the model where I need. Non-Prusa Edition Slic3r (1.3.1-dev) does not experience this issue.

* _Actual Results_
  Video clip can be seen on my twitter here: https://twitter.com/FunKing3d/status/1054815402960842758

#### STL/Config (.ZIP) where problem occurs
Any stl I try does the same thing.

GSheltonIII, 2018-11-06:
> We have noticed similar behaviour. With a bed defined as 1515x505, parts cannot be manually positioned. Changing the bed shape to 1515x722 results in expected behaviour, but 1515x721 exhibits the problem.

I can verify this to be true. I just tested it in my copy and experience the same issue.

Interesting that you can set it larger and it works.

bubnikv, 2019-01-08:
@enricoturri1966 would you please look into this issue as well? Thanks.

FidelCapo, 2019-01-09:
This issue should be fixed in Slic3r PE 1.42.0-alpha2.

closed 2019-01-09 by FidelCapo