prusa3d/PrusaSlicer #1375: Objects with multiple parts - outside bounding area

OmensArea51, 2018-10-30:
### Version
_Version of Slic3r used goes here_Slic3r Pursa Edition 

_Use `About->About Slic3r` for release versions_1.41.0+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Win

### Behavior
* _Describe the problem_
When I import one stl and then add a sceond stl as a sub part the new combined parts will not slice.  One of the parts is green while the sceond part is blue.  Blue means it thinks it is outside of the print volume.  I know this is not the case because both parts imported by them selves fit in the print volume with out any problems.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_

To slice and generate G-Code

* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_

New request

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

I can not the file is to big.

FidelCapo, 2018-10-31:
Thank you for report. This issue should be fixed in 1.41.1-beta. Can you confirm this issue is fixed with 1.41.1 release? Thank you

OmensArea51, 2018-10-31:
I will have to download it and try it.

Thanks for the update.

Jene Omens
President, Florida Sculptors Guild
Artist-Sculptor-Creative

407-227-7692
www.jeneomens.com

OmensArea51, 2018-11-02:
Yes this version fixed the issue.  Thanks again.
Jene Omens
President, Florida Sculptors Guild
Artist-Sculptor-Creative

407-227-7692
www.jeneomens.com

FidelCapo, 2018-11-02:
Glad to hear it. I closing this issue.

closed 2018-11-02 by FidelCapo