prusa3d/PrusaSlicer #1377: Object outside of build volume

VjyedSC0UFMc8z, 2018-10-30:
### Version
1.41.0+

### Operating system type + version
macOS Sierra 10.12.6 (16G1114) 

### Behavior
I designed a 2 part MMU print for PLA in black a white.
Both stls are <200mm. The white STL is 191.2 x 191.2 x 5 and the black is 142.2 x 142.2 x 5. Both should easily fit on the build plate by themself.
When I follow the standard procedure for MMU slicing and place the larger white object and add the black with "Load Part" the black object is now shown as too large for the build volume. I need to shrink the whole assembly to 130mm square before Slic3r will slice it. Why?

If I combine both in meshmixer into an AMF (which should be the industry standard) I can print at full size.

_Is this a new feature request?_
no 

#### STL/Config (.ZIP) where problem occurs

Using Slic3r PE defaults
STLs and AMF attached.
[MMU multipart problem.zip](https://github.com/prusa3d/Slic3r/files/2530597/MMU.multipart.problem.zip)

FidelCapo, 2018-10-31:
Thank you for report. This issue should be fixed in 1.41.1-beta. Can you try this model with 1.41.1 release? Thank you

bubnikv, 2018-12-12:
It is supposed to be fixed a long time ago. Closing.

closed 2018-12-12 by bubnikv