prusa3d/PrusaSlicer #1378: Holes in design are created and then covered over.

ogg1e, 2018-10-30:
### Version
1.41.1

### Operating system type + version
OSX v10.14

### Behavior
I have 5 holes on the top of my design. When it prints, it prints the holes correctly.  However, as it continues to print the rest of the layers, it prints over the 5 holes.  If I view it in preview, it does show the holes are covered. If I use cura, it correctly slices it in the layer view.

I'm using the default settings in slicr for my mk2s. 

3D view with holes:
[image]
Preview with missing holes:
[image]

#### STL/Config (.ZIP) where problem occurs
[Archive.zip](https://github.com/prusa3d/Slic3r/files/2532085/Archive.zip)

wabbitguy, 2018-10-31:
I checked the design and see what you mean. I also noticed there was 298 errors in the STL file so I fixed it with NetFabb and now it works correctly when I slice it.

See if the fixed version slices correctly for you.

[stl_fixed.zip](https://github.com/prusa3d/Slic3r/files/2532629/stl_fixed.zip)

Mel

ogg1e, 2018-10-31:
interesting... I used sketchup to generate the stl. There weren't any errors in sketchup when I created the design.    I don't own netfabb, so what do I do in the future when this happens again?

bubnikv, 2018-10-31:
Sketchup is notorious for producing bad STLs.

You may try to fix your STL using the Netfabb service running by Microsoft.
By the way, it is the same service, that Slic3r uses when running on
Windows 10.

https://tools3d.azurewebsites.net/

ogg1e, 2018-10-31:
Man. That's disheartening. I've spent all this time learning sketchup only to find out it's crap at generating STLs?? Sigh... 

Thanks for the help

closed 2018-10-31 by ogg1e

ogg1e, 2018-10-31:
I tried using that tools3d service to fix it, and it generated a 3mf file (never used that before). But when I load it and slice it in slicr, it's still not correct.

FidelCapo, 2018-10-31:
You can use a netfabb on web with Mac. https://service.netfabb.com/

FidelCapo, 2018-10-31:
I tried to fix the object in a slicer on win using Netfabb and Netfabb web app. In both cases, it seems to me that the object is being sliced correctly. I'm attaching a repaired stl from netfabb web app.

[Pi1541-oggie-top-large-lcd_fixed.stl.zip](https://github.com/prusa3d/Slic3r/files/2534351/Pi1541-oggie-top-large-lcd_fixed.stl.zip)

ogg1e, 2018-10-31:
Thanks @FidelCapo . I redid it and it worked. I exported again from sketchup, and reuploaded to netfabb webapp and it worked for me. I guess I'll be using that app more often.

FidelCapo, 2018-10-31:
You will probably have to. My personal recommendation is Fusion 360.