prusa3d/PrusaSlicer #1379: Crashes when overriding top or bottom perimeters in settings modal.

shendrick, 2018-10-31:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_
Version 1.41.1+

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
macOS Mojave 10.14

### Behavior
* _Describe the problem_
Slic3r crashes when changing top or bottom layer numbers in the object settings modal.
* _Steps needed to reproduce the problem_
Load a model.
Open the settings modal.
Choose to override the print settings top or bottom layer numbers. 
Change to anything else.
Click Ok.
It crashes. :(
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
It shouldn't crash.
* _Actual Results_
It crashes.
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Here's the log...

`
Process:               perl5.26.1 [27904]
Path:                  /Applications/Slic3r.app/Contents/MacOS/perl5.26.1
Identifier:            com.prusa3d.slic3r/
Version:               Slic3r 1.41.1+ (1.41.1+)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           perl5.26.1 [27904]
User ID:               501

Date/Time:             2018-10-30 22:27:44.489 -0700
OS Version:            Mac OS X 10.14 (18A391)
Report Version:        12
Anonymous UUID:        B57423A4-D7F7-B65B-67D6-A8E3C7981466

Sleep/Wake UUID:       6417E283-6C15-4065-920E-807FD2598FDF

Time Awake Since Boot: 220000 seconds
Time Since Wake:       5200 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [27904]

VM Regions Near 0:
[... 530 lines cut ...]
===========                     =======  ======= 
TOTAL                              1.0G     1177 
TOTAL, minus reserved VM space     1.0G     1177 

Model: MacBookPro10,1, BootROM MBP101.00F6.B00, 4 processors, Intel Core i7, 2.7 GHz, 16 GB, SMC 2.3f36
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Graphics: NVIDIA GeForce GT 650M, NVIDIA GeForce GT 650M, PCIe
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xEF), Broadcom BCM43xx 1.0 (7.21.190.32 AirPortDriverBrcm4360-1302)
Bluetooth: Version 6.0.8f6, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM768E, 751.28 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4
`

FidelCapo, 2018-10-31:
Thank you for report. Unfortunately, I'm not able to reproduce this issue. Can you send me 3MF file with this issue? Thank you.

FidelCapo, 2018-10-31:
Do you have background processing enabled?

bubnikv, 2018-12-12:
No response. Hopefully this issue is fixed in 1.42 alpha, as we have reworked the background processing logic. Closing.

closed 2018-12-12 by bubnikv