prusa3d/PrusaSlicer #1380: Estimated time mishmash

ondratu, 2018-10-31:
### Version
1.41.1+linux64 (appimage) from web

### Operating system type + version
Debian Buster

### Behavior
Estimated printing time with bad values like -2147483648s or milon days etc.

[image]

#### STL/Config (.ZIP) where problem occurs
[xyzCalibration_cube.zip](https://github.com/prusa3d/Slic3r/files/2534109/xyzCalibration_cube.zip)

[config.zip](https://github.com/prusa3d/Slic3r/files/2534110/config.zip)

FidelCapo, 2018-12-13:
Thank you for report. I can reproduce wrong print estimated time in silent mode.

FidelCapo, 2019-04-15:
Created JIRA issue 903

bubnikv, 2019-05-14:
This will be fixed in PrusaSlicer 2.0.0-rc. Closing.

closed 2019-05-14 by bubnikv

guestisp, 2019-05-14:
Great, it was related to mine. I'll close mine too.
https://github.com/prusa3d/PrusaSlicer/issues/2144

jonnykl, 2020-04-28:
I'm still having this issue.

Version: 2.2.0+linux-x64
OS: Ubuntu 18.04.4

Since a few days I'm also sometimes getting a print time of -2147483648s (or less often other bad values). Some configurations/STLs will never output a reasonable time, on some other configurations/STLs it works after removing/adding the object or moving it around and then back to the previous location.

[config_and_stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4547705/config_and_stl.zip)