prusa3d/PrusaSlicer #1382: G-code Flavor for i3 MK2? Marlin vs RepRap/Sprinter?

chadwixk, 2018-10-31:
### Version
1.41.1 Win64

### Operating system type + version
Windows 10 Pro

### Behavior
* _Describe the problem_

Printer settings for the Original Prusa i3 MK2 has the G-code flavor selected as Marlin and it has the Locked Icon next to it, implying that is the default value. But when I compared this to my older printer settings file, it had RepRap/Sprinter. Also, looking at the tool tip for G-code Flavor states that RepRap/Sprinter is the default.

Which is correct?

* _Steps needed to reproduce the problem_
  Select the Original Prusa i3 MK2 printer setting

* _Expected Results_
G-code Flavor to be RepRap/Sprinter

* _Actual Results_
GCode Flavor is Marlin
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_
No

chadwixk, 2018-10-31:
Thanks @bubnikv for pointing me to that. But I'm still not sure on which one to choose based on the comment below. 

"_**If you are running Marlin**, please make sure to switch Slic3r to a new "Marlin" firmware flavor to get a correct print time estimate_"

If you are running Marlin....where? Is this referring to the firmware in the printer itself? If so, how do I know if my MK2S, running firmware 3.1.0 is "Marlin" or RepRap?

bubnikv, 2018-10-31:
>  Is this referring to the firmware in the printer itself?

Yes

Prusa3D firmware is derived from Marlin, Marlin settings is correct.
Indeed, this is what comes online with the Prusa3D system profiles.

closed 2018-10-31 by chadwixk