prusa3d/PrusaSlicer #1384: feature request - overwrite default profiles

photog0411, 2018-11-03:
Hi guys. The default prusa print profiles work for me almost perfectly so I use them, but there are a few settings I would like to tweak in them. It would be nice if you could save over the default profiles (right now it doesn't let you). I realize that you can copy them and save your own profiles but I don't really need double the profiles to select from it just clutters the list. To make this request even better, you could add a button to reset the default profiles back to the original settings in case we muck something up

marhar, 2018-11-04:
Just IMHO, I like the current behavior.  We have a shared printer and it's very excellent not to worry that something has been changed by someone else. Or that I have accidentally messed something up!

photog0411, 2018-11-04:
This is what options are for. :)

photog0411, 2018-11-04:
@marhar this is what options and settings are for :)

bubnikv, 2018-12-13:
Sorry, but we will not allow overwriting the system profiles. The current behavior is on purpose: To make Slic3r a solid experience. 

Please understand, that we employ an army of support stuff. If we allow the system profiles to be overwritten, or if we allow them to be hidden, many people will mess up and this will make their communication with our support difficult.

Closing.

closed 2018-12-13 by bubnikv

photog0411, 2018-12-13:
@bubnikv I understand. Perhaps what I'm experiencing is more of a bug? See below screenshot. It appears the default profiles are duplicated and that's why I'm having this issue.

[image]

bubnikv, 2018-12-13:
Please zip and post your Slic3rPE config directory here. We will look into it. It may be a result of some profile update going wrong. In my opinion, you should be able to delete the user profiles from the user interface.