prusa3d/PrusaSlicer #1386: Soluble interface generation issues

Kolous, 2018-11-04:
### Version
1.41.1+win64

### Operating system type + version
Windows 7 x64

### Behavior

Couple floating lines of the wipe tower:
[image]

Extra soluable layers in center hexagon. 
Should be:

- soluable
- solid

Instead is generated:

- soluable
- solid
- soluable
- solid

[image]
[image]
[image]
[image]

#### STL/Config (.ZIP) where problem occurs
[a.zip](https://github.com/prusa3d/Slic3r/files/2546113/a.zip)

bubnikv, 2018-11-05:
Please see https://github.com/prusa3d/Slic3r/issues/1355
It should be a FAQ

Kolous, 2018-11-05:
Ok, line clarified. The second issue remains ;-)

bubnikv, 2019-01-29:
I suppose the inner wall is not completely straight, there is a kink, which slic3r evaluates as in a need of support, so it puts the support there.

It is an unusually difficult job to filter out such tiny spots that seem to require a support while not missing tips of sharp objects, which do require supports.

bubnikv, 2021-02-24:
The second issue is fixed with 6bdd42b1cfd34fe4171d24c5c59260d1aa9b12bf and it is a duplicate of #4570

    Disabled a suspicious piece of legacy code inherited from Slic3r,
    which tries to fill in spots inside support interfaces. The code was
    too aggressive and it was doing more harm than good.

closed 2021-02-24 by bubnikv