prusa3d/PrusaSlicer #1387: Holes in a hole though my object, dear Liza...

RobMeades, 2018-11-04:
### Version
1.41.1-rc+win63

### Operating system type + version
Windows 10, 64 bit.

### Behavior
* Original STL, available [here](https://1drv.ms/u/s!AqUHa_qo4gdB8gYQUtDa6Gz6c70j), shows nice hole through object, as intended:

[image]

However, sliced version shows that the cylinder is not complete, there is a cutting through the bottom (and top) surfaces, exposed through to the in-fill:

[image]

* To see the proble, slice the original STL file, doesn't seem to matter what the settings are but I am using a PRUSA MK2/S at "0.15 mm OPTIMAL" settings; the `File -> Export Config` can be found here: https://1drv.ms/u/s!AqUHa_qo4gdB8gevadnhwdYQT38N 

Is this something I can resolve in my settings somehow?

lukasmatena, 2018-11-05:
I don't understand what else should Slic3r do. Printing always separates the object into distinct and FLAT layers. The layer just below the opening ends up as a flat plane. The layer immediately after already cuts through the opening and a rectangle emerges. And obviously you see the flat layer below through this rectangle. This effect will always be present, more so with higher layer height. You can get rid of it by printing the hole vertically (which may not be possible, depending on the rest of the object).

In case I misunderstood the problem, please clarify what behavior you expected.

swilkens, 2018-11-05:
The geometry is simply too small to be represented by 0.15 layer heights in the accuracy that you are looking for, if you want more layers to detail that section you might look into using variable layer heights around the areas where this detail is needed.

There is no hole, the red color you are seeing is a solid infill of the layer below. The pictures below illustrate what I mean. This is what happens when a round opening is represented by slices.

[image]

That .STL is massive by the way, you might reduce the mesh detail

RobMeades, 2018-11-05:
Thanks for the swift responses.  I now see that the red area is a view of in-fill, my reason for asking about this is that the top and bottom of the hole seem to be particularly bad areas in my Prusa print.  Here's the printed result in 0.15 mm resolution:

[image]

...and if I increase the resolution to 0.05 mm it is still there:

[image]

I would say that this is simply a fact of life, gravity exists for FDM printers, however I sent the STL off to be printed on an extremely expensive (still FDM) Fortus 450MC printer in 0.127 mm resolution and it came out like this:

[image]

Despite the comparatively poor resolution on the expensive printer there is little, if any, distortion at the tops and bottoms of the hole.  They are printing in ASA (my target material) while I'm still printing in PLA, the other difference is that they probably sliced it in something else.  How did they manage it?  Did they maybe rotate the print and so that the hole was printed vertically?

swilkens, 2018-11-06:
Material properties, flow rate, cooling, support material under the overhangs... You are essentially extruding plastic into thin air there, there is nothing preventing the material from drooping down but its own stiffness. Tuning that will require you to play with the printer configuration.

From your pictures, we can see that both your private prints and the prints on the 450MC are printed in the same orientation. That said, it is impressive how little droop their print shows on the first layers. Probably they used some form of support material, preventing a droop.

I can also see some rippling on your 0.15 mm layer height print, suggesting the printer is mechanically oscillating after making that hole geometry - which isn't surprising given the abrupt direction changes. You might look into acceleration, jerk and speed tuning for your printer to better print such small geometry in a better quality.

RobMeades, 2018-11-06:
Agreed this is not a slic3r problem.  FYI, my original prints did include support structures and the 0.05 mm layer height was a lot less ringy in general.  I wonder how the Fortus 450MC did it though?

closed 2018-11-06 by RobMeades