prusa3d/PrusaSlicer #1388: Using printer config "MK3 Speed Profile PLA" disables support generation.

marhar, 2018-11-04:
### Version
1.40.1+

### Operating system type + version
MacOS 10.14 Mojave

### Behavior
"MK3 Speed Profile PLA" disables support generation. https://github.com/eoprede/prusa_profiles

* _Steps needed to reproduce the problem_
* Set printer to Original Prusa i3 MK3
* Slice the model with Support: Everywhere
* observe Preview, see support generated
* set printer to MK3 Speed PLA Profile
* Slice printer
* observe preview, no supports!
[image]
[image]

### Expected Behavior

* Supports are generated
* If problem, get message regarding configuration

### test file
[chassis_small.stl.gz](https://github.com/prusa3d/Slic3r/files/2546425/chassis_small.stl.gz)

FidelCapo, 2018-12-13:
Unfortunately, I am unable to reproduce this issue. Can you please try that bug is fixed in the 1.42.0-alpha version that will be released soon? Thank you

thisiskeithb, 2019-01-27:
I'm running into this issue with two different computers running macOS Mojave 10.14.2 and Slic3r 1.42.0-alpha3+ and using [Chris Warkocki's profiles]( https://github.com/eoprede/prusa_profiles) mentioned above. As soon as I switch back to a default/built-in profile, supports work fine.

bubnikv, 2019-01-29:
Would you please zip and attach your Slic3rPE (or Slic3rPE-alpha) folder with your presets, so we can reproduce the issue?

bubnikv, 2019-04-13:
no response, closing

closed 2019-04-13 by bubnikv