prusa3d/PrusaSlicer #1389: Thin wall detection reversed

ArjanStr, 2018-11-05:
### Version
1.41.1+win64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows10
### Behavior
If thin wall detection is checked, slicer does NOT detect thin walls.
The selection is reversed.

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lostcontrol, 2018-11-14:
Ah yes, seems you're right. Good catch!!!

matthew-humphrey, 2018-11-14:
Wow, this is a big deal for folks using this for structural parts like those for 3d printers. Any ETA on fix or info on which commit broke this?

bubnikv, 2018-11-15:
@FidelCapo would you please confirm this? Thanks.

FidelCapo, 2018-11-15:
I am not able to reproduce this error. I've tested the "detect thin perimeters" feature on the attached object, and it seems to me that the function works correctly.

Model sliced with "detect thin perimeters" feature disabled
[image]

Model sliced with "detect thin perimeters" feature enabled
[image]

Can you please send me a 3MF file where the issue is showing?

wavexx, 2018-11-16:
I also have some weird issues with "thin wall" detection. As the OP, I mostly do mechanical parts, and design features with exact perimeter sizes in mind.

The option works *mostly* as intended. A 0.4mm wall will not be printed with a 0.4mm nozzle unless the option is switched on.

However, there are definitely some corner cases where thin wall detection actually *prevents* some thin (and not so thin) features to be printed. I'm trying to find the specific cases where I had this problem in the past and post some models.

bubnikv, 2018-12-13:
@lostcontrol The first image shows the effects of the thin walls detection to me indeed, so the switch is certainly correct, only the "detect thin walls" does not work optimally. Your model is a good test object for @supermerill , who is currently reworking the gap fill algorithm.

supermerill, 2018-12-13:
> gap fill algorithm

It's in the review process in slic3r upstream. If/when merged, I will do the same here.

lostcontrol, 2018-12-14:
@bubnikv ok, makes sense. @supermerill [here](https://github.com/prusa3d/Slic3r/files/2678860/toad90.stl.gz) is the STL file. I use 3 shells and I think my layer width should be at 0.45mm. I can export a AMF or 3MF file if you want.

supermerill, 2018-12-14:
@ lostcontrol  I don't know why you give me this stl but i can slice it and i can have thin walls.
[image]

lostcontrol, 2018-12-14:
@supermerill because @bubnikv said a few comments that:

> @lostcontrol The first image shows the effects of the thin walls detection to me indeed, so the switch is certainly correct, only the "detect thin walls" does not work optimally. Your model is a good test object for @supermerill , who is currently reworking the gap fill algorithm.

That's why I attached that model. Have a look at the [same model but now saved as AMF](https://github.com/prusa3d/Slic3r/files/2681558/toad90.zip.amf.gz) with my settings which should show the problem. At least it does look weird for me with Slic3r 1.14.1:

[image]

supermerill, 2018-12-14:
On my fork, coming into slic3rPE next year: 
[image]

lostcontrol, 2018-12-16:
@supermerill looks great, looking forward to it :+1: But why do you only have 1 shell? How does it look like with 3 shells?

closed 2026-09-29 by prusaslicer-bot[bot]