prusa3d/PrusaSlicer #1393: Improvement of underside face that are being bridged

BAKMS, 2018-11-07:
### Version
1.41.1

### Operating system type + version
Windows 10

### Behavior
Underside faces are not smooth or very spaghetti-ed faced.
Requesting assistance to improve the underside faces that are being bridged.

_Is this a new feature request?_
Yes

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

supermerill, 2018-11-07:
Can you export the platter after slicing? Go to platter -> export as 3mf, then link the file here.

A picture of the problem will be also welcome.

BAKMS, 2018-11-07:
[image]
[image]

Sorry I do not access to the .3mf at this time.
I have attached pictues

supermerill, 2018-11-08:
Thanks.

These are surfaces that are on supports?

Difficult to say, with only half the information. I will wait for the 3mf.

BAKMS, 2018-11-08:
yes the surfaces are directly on the supports

bubnikv, 2021-03-18:
To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio. 

[image]

Implemented with ceea9de8b8b1d4b35bdd1180051033680c405ba1 f01f02154c617cc24835475da419ed6322f2ceec

closed 2021-03-18 by bubnikv