prusa3d/PrusaSlicer #1394: Feature request: apply scale to all objects

zbrozek, 2018-11-08:
To account for annealing shrinkage I tend to scale all parts on a plate by the same values. Doing this one part at a time is tedious and I'm prone to missing some axes on some parts.

It would be lovely if I could scale a whole plate. It would be even more lovely if I could set scale factors as part of a profile, so I stop forgetting to do it altogether.

Super bonus extra brownie points if scaling weren't done in percent, but rather as a number relative to one; e.g., 105% becomes 1.05.

faboaic, 2018-11-08:
... and select multiple objects to be able to move all at once :-)

lukasmatena, 2018-11-09:
Thanks for the suggestion. Multiple object selection is one of the new features of the upcoming release 1.42 and it already supports both batch moving and scaling (in our development branch). That should make the job easier.

Just out of curiosity - what's the significant difference between 105 % and 1.05? It seems to be nothing but a cosmetic thing to me.

BETLOG, 2018-11-09:
+1 for this feature. Specifically *selection set* transformation, not necessarily 'all' objects.

Also: a workaround is export all as stl, and reimport makes them a single object. Which is easily scaled.

zbrozek, 2018-11-09:
@lukasmatena - why 1.05 and not 105%? It's because I do math like, "desired size / measured size" and get a near-one value. And other software (e.g., Inkscape or Fusion 360) also uses fractional values. Net result is that it's a multiplication step I always forget and end up with comically proportioned parts. And as far as I know, there's no undo button, so I delete the part and do it over again.

@BETLOG brilliant workaround. I didn't realize it was for the whole plate, not just whatever the selected part was. My personal workaround so far has been to just use a python script to rescale parts from the command line. But that has its own problems. A profile setting would be amazing.

BETLOG, 2018-11-10:
It operates on plate or part depending on where/how you invoke it. Button next to 'export gcode' is the entire plate.

BETLOG, 2018-11-10:
Also, if you mess up the scale, i'm pretty sure that repeating the scale command brings up the current scale in the field, so just reset it to 100

bubnikv, 2018-12-13:
Group selection and group operations (translate, rotate, scale) will be implemented in the Slic3r 1.42 alpha Christmas release.
Closing.

closed 2018-12-13 by bubnikv