prusa3d/PrusaSlicer #1395: Preset Dropdown selectors do not have a maximum width limit on Ubuntu

simonkuehling, 2018-11-08:
### Version
1.41.1+linux64

### Operating system type + version
Ubuntu 16.04 LTS

### Behavior
* When often loading a configuration from another Gcode file (File --> Load Config --> "myprint.gcode"), the preset names shown in the Plater dropdown menus on the right are getting very long (nested names with each old name in parantheses). Problem is that on Ubuntu the dropdown selector always grows indefinitely to display the full preset name. The 3D view on the left may become cut off completely.
* Screenshot
[image]
* I would suggest to either define a maximum width for the dropdown controls or to add a method to manually slide/resize the size of the 3D preview <--> presets selectors ratio

bubnikv, 2018-11-08:
We should make the width of the right column fixed.

simonkuehling, 2018-11-08:
That would be an easy fix I guess, sounds good to me 👍

commit 48a94eba (2018-12-14): Fixed scroll the Object List to selected item on OSX & GTK

+ fixed #1395 (maximum PresetComboBox's width limit on Ubuntu)
closed 2018-12-14 by YuSanka