prusa3d/PrusaSlicer #1397: custom gcode fields unusable

BETLOG, 2018-11-09:
### Version
Slic3rPE-1.41.1-beta+linux64-full-201809261049.AppImage
and
Slic3rPE-1.41.1+linux64-full-201810261150.AppImage

### Operating system type + version
Kubuntu 14.04

### Behavior
* _Describe the problem_
Keyboard based movement of the cursor instantly resets any selections.
Attempting to edit text that is below the initial visible area instantly resets the visible portion but leaves the cursor where the user put it, and offscreen.
Keyboard selections are deselected.

* _Steps needed to reproduce the problem_
Place the cursor inside a gcode text box. Printer settings usually has large blocks of code, so thats a decent example, but afaik they all do this:

1) select some text with:
 home -> shift end
The selection appears briefly, and is then deselected. The cursor does move to EOL.

2) keyboard PGDN in a field that has many lines of text.
The text does not scroll down (it does, but instantly resets), the cursor does despite being offscreen.

3) keyboard downarrow in a field that has many lines of text.
The cursor will move but the field of view will constantly jump back to the top.

* _Expected Results_
Text should be editable inside slic3rPE. Right now I have to cut/paste the entire field into an external editor before I can do anything.
This glitch has existed and become worse for the last few months (that I can remember), and is now so bad it is unusable.

* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_
I'll upload a video if requested, but unless I'm the only one experiencing this I'm sure the problem will be self apparent to anyone else experiencing it. I'm surprised that there isn't already an issue registered from months back.

BETLOG, 2018-11-13:
Opening the profile directly in a text editor would probably be the most direct, but the syntax and formatting requires extra attention to detail if you do it that way.
I'm wondering why there arent hundreds of people reporting it, maybe OS related?

JesusFreke, 2018-11-13:
Yeah, you have to deal with the encoded newlines of course. It was pretty straightforward to do a regexp search and replace for that though.

I forgot to mention, I'm also on ubuntu. version 17.10. So could definitely be a ubuntu/linux thing.

telanoc, 2018-11-22:
Also happens here on Slackware 14.2.  Thank goodness for cut and paste.  It would really be nice if this was fixed, though.  I'm going to a friend's house tomorrow and he's got Slic3rPE on a Windows machine.  I'll see if it happens to him as well and update.

telanoc, 2018-11-23:
As a followup, this doesn't appear to happen on my friends machine, under Windows.

BETLOG, 2018-11-29:
And now it seems to be becasue I have < and > in a REMARK.
I often copy and paste the syntax example from the Marlin gcode page so I remember what i'm doing in my Slic3r custom g-code fields... it's been a while since I had a < or > in there, but I'm sure I would have pasted them before, and don't recall needing to fix this before.
Surely anyhting processing rems in gcode would be able to ignore redirect symbols in a remark?

[image]
[image]
[case-battery80x90x4mm-001__LID__60-60C_200-200C__0.2mm-0.1-0.3mm__2x0.4-0-0mm-40mms__20%triangles__20181129-105028.gcode.tmp.zip](https://github.com/prusa3d/Slic3r/files/2627146/case-battery80x90x4mm-001__LID__60-60C_200-200C__0.2mm-0.1-0.3mm__2x0.4-0-0mm-40mms__20.triangles__20181129-105028.gcode.tmp.zip)

bubnikv, 2018-11-29:
Most likely your editor generated some unicode (non latin-7) characters.

bubnikv, 2018-11-29:
sorry, forget my previous comment.

bubnikv, 2018-12-13:
The upcoming Slic3r 1.42 alpha will only send the update events when the text edit field loses focus. This should fix all the UI issues you are guys experiencing.

In regard to the [keyword] stuff inside the G-code comments, sorry, we will not change the current behavior. Please replace [] with something, which does not offend the G-code preprocessor.

Closing.

closed 2018-12-13 by bubnikv