prusa3d/PrusaSlicer #1399: [Feature request] Set minimum number of bottom layers

jan-gerard, 2018-11-10:
### Version
1.41.1+win64

### Operating system type + version
Win7 SP1

### Behavior
Some models have a rather thin bottom layer. Without changing the stl model file I would like to be able to set a minimum number of bottom layers, like 2.

The printing of the 1st layer does not always result in a contiguous surface, but rather lines that touch but are not completely molten together. A simple solution is to put a 2nd layer on top of it, regardless of the original thickness in the model file, with the lines perpendicular to the 1st one. 

Or course I can modify the stl file, but basically I want a surface as thin as possible but with at least 2 layers. And I think it should be up to the printing process to determine the thickness of that, based on the layer height settings I choose. And therefore, this information should not go in the slt model file, but the printer settings should take care of that.

And therefore, I think it would be a desirable option in Slic3r to set the minimum bottom layer number.
I think this is a new feature request, as I could not find a related topic.

supermerill, 2018-11-12:
If i understand, you have a volume with a height of 0.2.
As you set your layer height also to 0.2, it's only 1 slice.

You want at least 2 slices, so you want the slic3r to change the layer height to 0.1 for this volume?

You can do that with the layer editing:
 * go to your printer setting and set the min layer height to 0.1
 * layer edit your object to have the min at this surface.

You can also use the tool for user-defined layer height (for a sharp transition):
 * right-clic on the object
 * got ot the layers tab
 * put in the first line the height of your surface (like 0)
 * put in the second cell the top height of your surface (like 0.2)
* put the modified layer height in the thrid one (like 0.1)

Done!

> the printer settings should take care of that.

Everyone have different expectations.

jan-gerard, 2018-11-13:
Thank you for your help. There are many more functions than I had found so far. I'll continue exploring.
I will try to find what is the minimum layer height (and hence horizontal thickness as well) such that I can print objects that have a flat surface with 2 layers at the bottom. 
What I was initially looking for is a way to force the slicer to make at least 2 layers on the bottom, regardless of the object thickness and layer height, so to make it print a thin layer of let's say 0.2 mm as two layers of 0.2 mm (if 0.2 mm is the minimum my printer can handle). It get's thicker than specified in the volume file, but at least it's got two layers and that's what I want to achieve.
I didn't find a way to do that yet in Slic3r.

supermerill, 2018-11-13:
>What I was initially looking for is a way to force the slicer to make at least 2 layers on the bottom, regardless of the object thickness and layer height, so to make it print a thin layer of let's say 0.2 mm as two layers of 0.2 mm

Not possible, as far as i know. Slic3r will only slice inside the given volume/object.

bubnikv, 2018-12-13:
Sorry, but we will not implement your request, certainly not this or the next year.

You are basically asking Slic3r to:
1) Analyze the object, and decide, whether the object is hollow? What if the object has some holes? What if the object is open, like a vase?
2) If the first condition is fulfilled, then to fill some areas with holes on  2nd and possible couple of more slices. Which holes?

This is a very involved feature, which basically modifies your input object's geometry. It is possible, but it would be difficult to implement, difficult to understand by the users, and there are lower hanging fruits to implement.

Closing.

closed 2018-12-13 by bubnikv