prusa3d/PrusaSlicer #1400: Question - Why was pillar removed in Prusa3D

JamesVerner, 2018-11-11:
Hey, I'm doing research on slic3ring programs and their related algorithms. I was wondering would anyone know whay Slic3r PE removed pillars? Thanks.

lukasmatena, 2018-11-11:
I was not on the project when that happened, but a quick search produced this: https://shop.prusa3d.com/forum/print-tips-archive--f86/slic3r-pillar-support-setting-t6080.html . Apparently they did not work reliably.

Is there some specific reason why you're asking, apart from the natural urge to know as much as possible?

bubnikv, 2018-11-12:
The pillars feature just took the "normal" support volumes and then trimmed
them by regular pillar geometry, not taking into account their stability. I
rewrote the original support generator some two years ago (the original
support generator was written in Perl and it had a major issue with fusing
to the object top surfaces), and I did not find it worthwhile to port the
IMHO non-stable pillars along the way.

JamesVerner, 2018-11-14:
Thanks for replying guys, I'm just curious to why exactly they were removed. @bubnikv I was wondering as well since you looked into the Perl. Would you remember how pillars actually got their shapes? I've been focusing on the sub generate pillar but I can't seem to track down how they form the actual shape?

bubnikv, 2018-12-13:
Sorry, I don't remember the details. I think the software calculated the support volumes and then it intersected the support volume with something like this (the cross section is square)

[image]

The main problem was, that the column could have been trimmed with the support volume to something, which was not stable.

Closing.