prusa3d/PrusaSlicer #1401: Profiles: ColorFabb XT and XT-CF20 "wrong" type

Blizzke, 2018-11-13:
These materials all contain `PET` in the filename because they inherit from the PET profile, which is not really correct.
Would it be possible to give them their own type so the proposed filename is correct? 
I work with PET, XT and XT-CF20, I have to make sure I don't forget to manually modify the filenames each time.

https://github.com/prusa3d/Slic3r/blob/4f676de1616c3942721e2431720c8de255be326b/resources/profiles/PrusaResearch.ini#L788

Blizzke, 2018-12-08:
Its wrong indeed. 
Been printing white XT so I didn't see before, but with black it turned out to be very matt. A sign of a way too high nozzle temperature. 

The colorfabb site says 240-260ºC, so the 260 (first layer) and 270 (next layers) are too hot. It also says 40-60ºC bed temperature. not the 90ª that the profile specifies. I've been experimenting with temperatures and it seems the 240-260 works for the nozzle fine but I can't get it to stick to my MK3 metal sheet at the advised bed temperatures.

Asked colorfabb and they said I shouldn't be afraid to increase the bed temp but the nozzle temp is too high.

MrMap, 2018-12-09:
I just found out that while naming the files ...PET... may seem confusing (as we choose the ColorFabb XT filament profile) it is not totally wrong: ColorFabb XT does indeed belong to the PET family, if I got it right.
https://en.wikipedia.org/wiki/Copolyester

Still, I think it would be more practical and less confusing to name the output files accordingly with the filament profile in use.

bubnikv, 2018-12-13:
Please post the profiles issue to https://github.com/prusa3d/slic3r-settings/issues
Jindrich will take care of it.

Closing here.

bubnikv, 2018-12-13:
@jindrichbenes  Please update the XT profiles.

closed 2018-12-13 by bubnikv

MrMap, 2018-12-13:
OK - I understand the issue is closed here.
I just want to add that after having printed more XT lately, actually the Prusa presets for XT produced the best results! Lower the bed temp and I got warp problems! So I will probably not file an issue on the slic3r-settings on github.

One problem I encountered before was due to a filament leak after nozzle change (I did not understand the problem then) - the XT preset initially got the blame for the mess, not fair!

jindrichbenes, 2018-12-14:
The XT does indeed belong to the PET family. It is labeled as PET in slicer settings because it is an ID for MMU1 regarding the ramming parameters. It being included in the file name is just a secondary use of that.
The higher temperatures were then again chosen based on test results.