prusa3d/PrusaSlicer #1404: Slicing leaves gap

jwvaughn, 2018-11-14:
### Version
Slic3r Prusa Edition Version 1.41.1+win64

### Operating system type + version
Win 10 version 1803

### Behavior
Slic3r PE seperate what should be one piece into two. The native version of Slic3r does the slice correctly.
STL file included

_Is this a new feature request?_ NO
[image]
[image]

STL/Config (.ZIP) where problem occurs
[STL and ini.zip](https://github.com/prusa3d/Slic3r/files/2578766/STL.and.ini.zip)

BerndJM, 2018-11-14:
Looks like there is something broken in your STL.
After sendig it through netfabb-basic's auto repair I get the attached file which renders fine ...

[Box-repaired.stl.zip](https://github.com/prusa3d/Slic3r/files/2578869/Box-repaired.stl.zip)

jwvaughn, 2018-11-14:
I have created the STL file several times with FreeCAD the results are always the same. I’ve also ran the error checking available in FreeCAD  every time and no errors.

 

While it is certainly possible there is an error from FreeCAD, why do the two version of Slic3r produce different results?  The Slic3r PE results using  your repaired file look identical to that from the generic Slic3r using my original STL file. 

 

This is the second time I’ve had different results between the two slicers suggesting to me that something is different in the slicing engines. In both cases, Slic3r PE generated a bad slice while the other was good.

thilo-adk, 2018-12-12:
Sorry, i forgotten the STL File. Here the FreeCAD STL Export File.

[Test.zip](https://github.com/prusa3d/Slic3r/files/2670633/Test.zip)

bubnikv, 2018-12-12:
Will it help, if you process the file with netfabb repair service?
https://tools3d.azurewebsites.net/

thilo-adk, 2018-12-13:
> Will it help, if you process the file with netfabb repair service? https://tools3d.azurewebsites.net/
> […](#)

There is a registration necessary. I found an alternative: [MeshLab](http://www.meshlab.net/) With MeshLab i can fix the Problem.

Thanks.

jwvaughn, 2018-12-13:
Slic3r PE has an option under the object tab to "Fix STL through Netfabb" which seems to work well and no regsitration required.  It worked to fix a couple of problems I had including the gap problem.

bubnikv, 2018-12-13:
Luckily, both the example STLs are sliced correctly by the upcoming Slic3r 1.42 alpha. Please stay tuned.

Closing.

closed 2018-12-13 by bubnikv