prusa3d/PrusaSlicer #1406: Could you detach your fork from slic3r

lsellens, 2018-11-14:
The code base has changed so drastically and forked code is not searchable in github. Could you guys detach your fork so we can use search please?

bubnikv, 2018-12-13:
Would you please more specific of which github tools are difficult to use due to the link between Slic3r and Slic3r PE?

I know the pull requests often land at the upstream Slic3r by mistake. What are the other github tools?

lsellens, 2018-12-13:
Just a simple code search can not be performed on a fork.

ctilley83, 2018-12-13:
It’s a pain and he’s right.  You can’t search forked repositories until the fork has more stars than the parent repository.

lsellens, 2018-12-13:
You can contact support and ask them to detach the fork.

bubnikv, 2019-04-15:
done

closed 2019-04-15 by bubnikv