prusa3d/PrusaSlicer #1407: Load Generic/Modifier

cjwescott, 2018-11-15:
### Version
Slic3r Prusa Edition 1.41.1+win64

### Operating system type + version
Windows 10

### Behavior
When attempting to position the modifier on a scaled part it is nearly impossible to position using the slide bars. The part displays in the window very small, probably based on the original scale. I typically import imperial and always scale up by 2540. 
File attached, however I believe this will behave the same for any imperial size part that is scaled 2540. 

[Pipe Hangers_1_2_FTG.zip](https://github.com/prusa3d/Slic3r/files/2583226/Pipe.Hangers_1_2_FTG.zip)

lukasmatena, 2018-11-15:
You are right that the behavior is incorrect. The modifier mesh manipulation is a pain in general, even without considering this bug. That's why it will be greatly improved in the next Slic3rPE release (it will be possible to edit the modifiers directly in the main scene). This scaling problem will be solved along the way.

As a temporary workaround I suggest to place the modifier mesh before scaling and leave the scaling as the last step in the process. This should correctly scale both the object and the modifier mesh. It is not anything particularly great, but it should do the trick.

bubnikv, 2018-12-13:
@lukasmatena described it well enough. We plan to release Slic3r 1.42 alpha soon. Closing.

closed 2018-12-13 by bubnikv