prusa3d/PrusaSlicer #1412: [Feature request]: Retraction settings on filament level

Boby71, 2018-11-18:
### Version
1.41.1

### Operating system type + version
Win10 64Bit

### Description
Currently, the settings for rectraction are located at the Printer Settings / Extruder screen.
In some cases it could be necessary to adjust retraction settings on filament level, e.g. when you have a very stringy filament (like some wood does). It' not really logical to set up a new printer just to be able to adjust the extraction for a certain filament.

It would make sense to be able to change the values here:
[image]

So either it would be a solution to...
1) move the extraction settings here OR
2) add additional extraction settings to that screen to override/correct the machine settings

Solution 2) would give the ullitmative flexibility, because retraction is depending on a) the machine (bowden/direct) as well as on b) the filament. If the same options would be provided on filament dialog called "Retraction overrides on filament level" this could be a solution for everyone. Thanks!

jonnieZG, 2018-12-06:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/480

bubnikv, 2018-12-06:
Duplicate of #480, closing.

closed 2018-12-06 by bubnikv