prusa3d/PrusaSlicer #1413: Machine Limits

jornbyte, 2018-11-18:
### Version
1.41.1

### Operating system type + version
Win10Pro

### Behavior
375/5000
Hello
I instaled the version 1.41.1 (Win10Pro).
Since then my printer is not working properly anymore.
This is due to the newly introduced Machine Limits.
The specifications I have adapted to my printer.
Homing is performed when the pressure is to start
is the Nozzel in the print bed.
This happens with every test. Everything is OK at 1.40.1.
Is there a way to turn off the machine limits?

Greetings from Saxony

supermerill, 2018-11-21:
hello

> Homing is performed when the pressure is to start
> is the Nozzel in the print bed.

i'm sorry but i don't understand.

The Machine limits are mainly use for the time estimation, and shouldn't prevent your printer from running.

The parameters that are going into the gcode are in the 'printer settings'-> 'custom G-code' -> 'Start g-code' section.

jornbyte, 2018-11-21:
635/5000
I have tested further. the machine limits are responsible. The Z axis is raised with step losses because the speed for pressure movements are no longer respected. When I turn off the machine limits, it's OK again.

M107
M190 S60; set bed temperature and wait for it to be reached
; M201 X5000 Y5000 Z50 E5000; sets maximum accelerations, mm / sec ^ 2
;M203 X400 Y400 Z2 E50; sets maximum feedrates, mm / sec
;M204 S15 T150; sets acceleration (S) and retracting acceleration (T)
; M205 X10 Y10 Z0.2 E2.5; sets the jerk limits, mm / sec
G28; home all axes -->Homing is performed OK
G1 Z20 F5000; lift nozzle --> Error step losses
M109 S225
G92 E0
G1 E35 F400
; Filament gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
G1 Z0.200 F7800.000; --> the Nozzel in the print bed

supermerill, 2018-11-21:
If your printer has some problem with this gcode
```
M107
M190 S60; set bed temperature and wait for it to be reached
G28; home all axes ;-->Homing is performed OK
G1 Z20 F5000; lift nozzle ;--> Error step losses
```

Then, the problem is in the firmware of your printer, or your current setting in the stepper drivers, or maybe a motor is broken.

if F5000 isn't erased by your firmawre to the max speed allowable.... reduce this number?

jornbyte, 2018-11-21:
**G-Code Error** 
M107
M190 S60; set bed temperature and wait for it to be reached
M201 X5000 Y5000 Z50 E5000; sets maximum accelerations, mm / sec ^ 2
M203 X400 Y400 Z2 E50; sets maximum feedrates, mm / sec
M204 S15 T150; sets acceleration (S) and retracting acceleration (T)
M205 X10 Y10 Z0.2 E2.5; sets the jerk limits, mm / sec
G28; home all axes -->Homing is performed OK
G1 Z20 F**5000**; lift nozzle --> Error step losses
M109 S225
G92 E0
G1 E35 F400
; Filament gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
G1 Z0.200 F7800.000; --> the Nozzel in the print bed

**G-Code without machine limits OK**
M107
G28; home all axes -->Homing is performed OK
G1 Z20 F**5000**; lift nozzle
M109 S225
G92 E0
G1 E35 F400
; Filament gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
G1 Z0.200 F7800.000

jornbyte, 2018-11-21:
one more note, I have been working slic3r1.1.7 with this program. The introduction of the machine limits makes not only me but also other user problems. does that turn off?

supermerill, 2018-11-21:
> The parameters that are going into the gcode are in the 'printer settings'-> 'custom G-code' -> 'Start g-code' section.

remove the unwanted lines (2->6)  
save the profile  
done!

jornbyte, 2018-11-21:
not done!
it is in the output G-Code :)

; generated by Slic3r Prusa Edition 1.41.2-rc + win64 on 2018-11-21 at 18:25:22

;

; external perimeters extrusion width = 0.45mm
; perimeter extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.24mm
; top infill extrusion width = 0.20mm
; first layer extrusion width = 0.40mm

M201 X5000 Y5000 Z50 E5000; sets maximum accelerations, mm / sec ^ 2
M203 X400 Y400 Z2 E50; sets maximum feedrates, mm / sec
M204 P15 R150 T15; sets acceleration (P, T) and retract acceleration (R), mm / sec ^ 2
M205 X10.00 Y10.00 Z0.40 E5.00; sets the jerk limits, mm / sec
M205 S0 T0; sets the minimum extruding and feed rate, mm / sec
M107
M190 S60; set bed temperature and wait for it to be reached
G28; home all axes
G1 Z20 F5000; lift nozzle
M109 S225
G92 E0
G1 E35 F400
; Filament gcode
,
,
,

supermerill, 2018-11-21:
I found the problem in the source code:

It prints it only for the marlin flavor.
Use the RepRap flavor for your printer, it's the same (but no limits / acceleration stuff inside the g-code).

jornbyte, 2018-11-21:
Thanks, it works

GwennG, 2018-12-11:
I was having the same issue when using Marlin flavor with Slic3r 1.41. I can confirm that after switching to Reprap flavor, the problem has disappeared.

Area5142, 2018-12-11:
> It prints it only for the marlin flavor.
> Use the RepRap flavor for your printer, it's the same (but no limits / acceleration stuff inside the g-code).
And time estimates for silent mode printing is missing (M73 Qx Sy).

Not a problem for me, as I never use it.

bubnikv, 2018-12-13:
This issue seems to be solved for everybody. Closing.

closed 2018-12-13 by bubnikv