prusa3d/PrusaSlicer #1414: Extruder clearance of 20mm is too small

timur-tabi, 2018-11-18:
In order for me to use this feature, I need to manually increase the value every time.  I set it to 30/30, but I suspect that even that is too small.

supermerill, 2018-11-21:
hello
I suppose by "In order for me to use this feature", you speak about "Complete individual objects"?
The bets way to do it is to measure the distance between nozzle and the (farest) border of the extruder (should be the right and the bottom in the prusa), then you put the values (+ some mm to be safe) int he fields, and save the new setting (with a new name if it's the default prusa one, as these one can't be erased). Next time, you just need to select this profile to ahve your good parameters

timur-tabi, 2018-11-21:
My point is, if we're going to have a Prusa version of Slic3r, the defaults should reflect the actual hardware.  I know I can save a custom profile with the correct values.  But I shouldn't have to.

supermerill, 2018-11-21:
> My point is, if we're going to have a Prusa version of Slic3r, the defaults should reflect the actual hardware. I know I can save a custom profile with the correct values. But I shouldn't have to.

Ok, you're right.

Maybe you should write that in the first post and modify a bit the title.
something like "the settings for prusa mk3 Extruder clearance are not the good ones". It will ensure people are able to understand what is your problem (I'm not a prusa employee, just a random internet guy).

For the future, when writing request / issue, please be very descriptive: what is my problem, what I want to change, what is my slic3rPE version, my printer, and send a 3mf file & screenshot of your plater if possible/needed.

bubnikv, 2020-02-06:
The current extruder clearance has been increased to 45mm.

[image]

Closing.

closed 2020-02-06 by bubnikv