prusa3d/PrusaSlicer #1416: Fan Header Selector?

HikariNoKitsune, 2018-11-19:
### Version
1.41.1-rc

### Operating system type + version
Win10 64bit

### Behavior
No ability to control WHICH fans are used...

_Is this a new feature request?_
Basically, unless there is another way to do this within Silc3r that I'm unaware about after 2+ hours of online searching.

Basically, I have a Duet WiFi... My default fan header for a cooling fan just isn't working anymore. It can't output enough power to turn on a fan... it'll output 12v, but it just doesn't have the current it seems to do anything.

SO I switched to the next controllable fan header which works just fine. 

Problem is, I can't get Slic3r to USE this fan header! It needs to be changed so that I can actually USE my part cooler, otherwise its useless... And I don't feel like coughing up 150+ to some company who thinks its slicer is so god like that it poops gold. . . 

Anyone know of how this can be accomplished? Or is it something that has to be coded into the slicer? I know I can just go into the GCODE and manually change the fan.. but why on earth take the time to go through thousands of lines of code, hunting for a single line countless times, and changing it to the correct fan header? That would take more time than I'd care to like...

HikariNoKitsune, 2018-11-19:
> You can set any fan to act as fan1 with a gcode. https://duet3d.dozuki.com/Wiki/Connecting_and_configuring_fans On Mon, 19 Nov 2018 18:32:51 +0000 (UTC), HikariNoKitsune <notifications@github.com> had a flock of green cheek conures squawk out:
> […](#)
> ### Version 1.41.1-rc ### Operating system type + version Win10 64bit ### Behavior No ability to control WHICH fans are used... _Is this a new feature request?_ Basically, unless there is another way to do this within Silc3r that I'm unaware about after 2+ hours of online searching. Basically, I have a Duet WiFi... My default fan header for a cooling fan just isn't working anymore. It can't output enough power to turn on a fan... it'll output 12v, but it just doesn't have the current it seems to do anything. SO I switched to the next controllable fan header which works just fine. Problem is, I can't get Slic3r to USE this fan header! It needs to be changed so that I can actually USE my part cooler, otherwise its useless... And I don't feel like coughing up 150+ to some company who thinks its slicer is so god like that it poops gold. . . Anyone know of how this can be accomplished? Or is it something that has to be coded into the slicer? I know I can just go into the GCODE and manually change the fan.. but why on earth take the time to go through thousands of lines of code, hunting for a single line countless times, and changing it to the correct fan header? That would take more time than I'd care to like...
> -- Stephen CONGRESS.SYS Corrupted: Re-boot Washington D.C. (Y/N)
> --- This email has been checked for viruses by AVG. https://www.avg.com

Yes I know. This is how the firmware is currently set up. 

The issue is, Prusa Slic3r does not allow you to SPECIFY which fan to use. It defaults to FAN0, as the wiki states. Simply changing the firmware to use FAN1 or FAN2 does not do 'anything' if the slicer in question does not allow you to specify which fan to use. As such, even though the board knows that there is a fan on say, FAN2, it will not use that fan as the default fan because in gcode, the slicer is calling on FAN0.

With Simplify3D you can specifically tell it "hey, my cooler fan is on FAN2, so use FAN2 when referencing a cooling fan"...

supermerill, 2018-11-21:
In the meantime, you can replace "M106 (S[0-9]+)" by "M106 P1 $1"
and "M107" by "M107 P1"

You write that in a script file.
Then, you paste the path to this script in printSettings ->output options -> Post-processing scripts.

HikariNoKitsune, 2018-11-23:
That's something I didn't know Slic3r could do, run a script. Or were you referring to just creating any ol script and running it afterwards?

supermerill, 2018-11-23:
> That's something I didn't know Slic3r could do, run a script

yes, you can create a script and slic3r run it (with the gcode filepath put at the end of the line) after exporting the gcode file.

bubnikv, 2018-12-13:
I like the idea of Simplify3D to do the basic script post processing using the REPLACE, PREPEND, APPEND, DELETE ... commands, which covers most of your needs.

https://forum.simplify3d.com/viewtopic.php?t=1959

Or should we just embed the unix sed tool?

closed 2026-09-29 by prusaslicer-bot[bot]