prusa3d/PrusaSlicer #1417: Bed heating

netzmark, 2018-11-21:
### Version
Version of Slic3r 1.41.1 Win64, 

### Operating system type + version
W7-64

### Behavior
- We declare some Filament settings FistLayer temperature
- it's creating "set bed temperature", next
- realizes start gcode section and after that 
- "set bed temperature and wait"
So if we have wipe put in start gcode it is made with cold extruder, after this the head stops and wait for temperature getting then continue. In this case wiping is not possible to realize and we extrude cold filament into the nozzle.
Why it is not made:
- get bed temperature 
- set bed temperature and wait
- start-gcode section
...

supermerill, 2018-11-21:
Hello

I'm sorry but i don't understand.
Can you explain the problem & link the problematic settings.

Silc3rPE has a bug?  
The default profile has a problem?

netzmark, 2018-11-21:
Hi

I think it is a bug of Slicer.
Look how the output file looks:

M107
M190 S60 ; set bed temperature and wait for it to be reached
M104 S210 ; set temperature
****start G-code begining
G1 Z5.0 F6000 ; move up 15mm at 6000mm/min
G92 E0 ; reset extruder length to zero
G1 X0.0 Y0.0 F1000 ; go to edge of print area
G1 X60 Z0 E9 F1000 ; start intro line
G1 X200 E45 F1000 ; finish intro line 21.5
G1 E44 F1000 ; retract 1mm
G1 X250 F4000 ; fast wipe away
G92 E0 ; reset extruder distance position
****start G-code ending
M109 S210 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0

At the begining "you" set
M190 S60 ; set bed temperature and wait for it to be reached - properly
M104 S210 ; set temperature - improperly
After start G-code section "you" set:
M109 S210 ; set temperature and wait for it to be reached

So the extruder temperature is not got before start G-code opeartion. In this case start G-code realize extruding and is forced to do this with cold extruder. After this only wait to get extruder required temperature.

It should be (in my oppinion) at the begining:
M107
M190 S60 ; set bed temperature and wait for it to be reached
M104 S210 ; set temperature - NOT SURE IF IT'S NEEDED IF M109 after it.
M109 S210 ; set temperature and wait for it to be reached - THIS IS MISSING HERE
****start G-code ending
etc.

Hope you understand now what I mean.

hevilp, 2018-11-22:
I can confirm it, the filament gcode should run after heating up hotend, but the gcode for filament is placed before heating it up, so my prime does not work

bubnikv, 2018-11-23:
As far as I remember, the process is following:

If you do not define the M104 / M140 codes in your start G-code, then:
1) Bed heating is initiated
2) Extruder heating is initiated and waited for the extruder to heat up
3) Bed heating is finished (wait for the bed to heat up)

If you use M140 in your custom G-code, the actions 1) and 3) will not be
generated.
If you use M104 in your custom G-code, the action 2) will not be generated.

You have a complete freedom to write the start G-code according to your
needs and then Slic3r will emit none of the M104 or M114 codes on its own.

hevilp, 2018-11-23:
;Prime the extruder ==> my custom gcode for filament
G92 E0.0 ; reset extruder distance position
G1 Z0.200 F1000.0 ; Go to Start Z position
G1 Y0.0 F1000.0 ; go to edge of print area
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E21.5 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position
M109 S210 ; set temperature and wait for it to be reached ==> this is from slic3r
and behind my custom gcode

No M104 / 140 is used in my custom gcode

netzmark, 2018-11-23:
You don’t understand.
In graphic interface there is initial extruder temperature and initial bed temperature. These parameters are placed at the begining  of the output code just after machine limit parameters definition.
Bed temp is set as „set and wait” but extruder is set as only „set”(missing”wait). Extruder „set and wait” is placed after both filament and start gcode sections. It’s mistake.

Od: bubnikv
Wysłano: piątek, 23 listopada 2018 10:01
Do: prusa3d/Slic3r
DW: netzmark; Author
Temat: Re: [prusa3d/Slic3r] Extruder heating (#1417)

As far as I remember, the process is following:

If you do not define the M104 / M140 codes in your start G-code, then:
1) Bed heating is initiated
2) Extruder heating is initiated and waited for the extruder to heat up
3) Bed heating is finished (wait for the bed to heat up)

If you use M140 in your custom G-code, the actions 1) and 3) will not be
generated.
If you use M104 in your custom G-code, the action 2) will not be generated.

You have a complete freedom to write the start G-code according to your
needs and then Slic3r will emit none of the M104 or M114 codes on its own.

netzmark, 2018-11-23:
Maybe expected but seems to be strange and could be improved :-).
Yes I have just put M109 S210 into start gcode...
I must check the Marlin behaviour what happens if to send sequence:
M104 S210 (put by Slic3r)
M109 without parameter (put into start gcode)
If this M109 will execute awaiting for previousli set temperature then nothing to improve.

hevilp, 2018-11-23:
What is the sense of put them manually Set Temperature? The is the sense of the slicer to Do it...

netzmark, 2018-11-23:
If bed can be heated before the start gcode with "wait" then hard to guess why not the extruder...
However if alone M109 in start gcode will fix the problem, let it be...

hevilp, 2018-11-26:
@bubnikv 

As I saw, the issue here were posted a few times since last 1-2 years.
Why the 'problem' is not fixed, when so many people are not happy with the actual behavior?

bubnikv, 2018-11-26:
@hevilp 
> Why the 'problem' is not fixed, when so many people are not happy with the actual behavior?

There is a very simple answer to your question: The day has 24 hours, and there are 509 issues open, every issue is the most important and annoying to somebody.

hevilp, 2018-11-26:
Okay, this answer is much better than "the actual behavior is fine and nice". I don't want to be harsh. I think to mark this issue as a bug is the right way and not to speak it away.

I know it is open source and I'm happy for every minute talented coders help to the project.

BTW: I saw at least 3 issues here are on that bug ;)

bubnikv, 2018-12-13:
> As I saw, the issue here were posted a few times since last 1-2 years.
> Why the 'problem' is not fixed, when so many people are not happy with the actual behavior?

Well, I think this "problem" was not fixed, because it is a problem for half of us, and it is an expected behavior for the other half. Now why would I break the application for me to make you happy?

The idea is to do whatever you need to do inside your custom start G-code before the bed heats up. Heating up the print bed may take a considerable time, so it is not waited for it implicitly. 

As noted by @supermerill 
> Just put the M104, M140, M190 and M109 in your start gcode if you want to manage them yourself 

Closing.

closed 2018-12-13 by bubnikv