prusa3d/PrusaSlicer #1422: Double-click .stl does not open model in OSX

duineuk, 2018-11-22:
### Version
_1.41.2-rc+_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_OSX Mojave 10.14.2 beta (build 18C48a)_

### Behavior
* _Double-click an stl file_
* _If Slic3r is associated with .stl, Slic3r will open_
* _Expected Result_ 
  * _Model is loaded and appears in Slic3r_
* _Actual Results_
  * _Model does not load / appear in Slic3r_

_Is this a new feature request?_
No.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

JohnOCFII, 2018-11-22:
I see the same behavior as @duineuk

codiac2600, 2018-11-22:
I also see this same issue.

FidelCapo, 2018-12-04:
I think this is same issue as #1239

ddavidebor, 2019-01-05:
@FidelCapo They're different issues.
#1239 is about how Slic3r interprets command line arguments.
This issue is about having Slic3r register itself as an application to be used to open .stl files (and may i suggest, also .3mf!) with the operative system.

Also, same for me:

[image]

bubnikv, 2019-01-10:
Fixed with c0df7f343ee28ccd96185a6599da8d848f692d35
Closing.

closed 2019-01-10 by bubnikv