prusa3d/PrusaSlicer #1424: Feature Request - Advanced first layer speed control

ingrimsch, 2018-11-23:
### Version
All Versions

### Operating system type + version
Any operating system
 
### Behavior
I would love to get more control over first layer print speeds. The ability to set advanced acceleration settings for the first layer was a great step in the right direction, but I often miss advanced speed setting in some print situations. An example:

I want to print a big model, but there are some screwholes or really sharp corners in the design/first layer. While I can easily print most of the first layer with 50mm/s or faster without running into adhesion problems (solid infill mostly) the perimeters and/or small perimeters would also be printed that fast, which results in adhesion problems. I can workaround that problem by manually reducing print speed, waiting for the perimeters to complete and then switching to 100% speed in the firmware settings. When printing multiple objects or having spaces in the first layer this workaround can get very time consuming, as individual areas are completed before the next one starts.

_Is this a new feature request?_
Yes. I havent touched any other Slicer since I switched to Slic3r PE some time ago, but as far as I can tell there is no other Slicer offering this today. I believe this would be a great addition for some print situations and wanted to ask this for discussion here.

### possible solution(s)
* In addition to first layer speed a "first layer perimeter speed" could be added (this would be my preferred solution)
* the small perimeter setting could also be considered in the first layer generation process

**Bonus request:** dynamic jerk settings for perimeters, first layer perimeters/first layer and maybe even bridges would also be awesome. This would be as useful as the advanced speed settings and be a great addition to them. It would help to get rid of ghosting/ringing on rigid machines with high acceleration/speed/jerk settings.

I love the features/additions you guys put into the Slic3r PE fork so far and it is by far the best Slicer IMHO. Please keep up the good work! :)

Peter

Area5142, 2018-11-27:
Have the same problem. Small screw holes loosens and get caught by nozzle and ruins 1. layer.
I too lowers the speed multiplier to around 40-50% to get corners and holes correct, but it takes ages to complete the solid infill...
If a percentage in first layer speed reduces reduces all other settings by that percentage this could be the solution, not limiting everything to 20-30mm/s.
I would like the travel speed to be reduced too as high travel speed sometime rips very small filament areas off the build plate with the filament string to nozzle (especially PETG).

supermerill, 2018-11-27:
> If a percentage in first layer speed reduces reduces all other settings by that percentage this could be the solution, not limiting everything to 20-30mm/s.

you can try right now.

>I would like the travel speed to be reduced too as high travel speed sometime rips very small filament areas off the build plate with the filament string to nozzle (especially PETG).

even with z-hop?

Area5142, 2018-11-27:
> 
> > If a percentage in first layer speed reduces reduces all other settings by that percentage this could be the solution, not limiting everything to 20-30mm/s.
> 
> you can try right now.

I'll try that.
I used to have percentage for first layer in my profiles, but reverted to standard profiles some time ago to follow PR recommendations. Maybe the standard PR profile should have first layer speed specified as percentage...
I think it is kind of more intuitive, like using the dial on the printer to slow speed down.

> > I would like the travel speed to be reduced too as high travel speed sometime rips very small filament areas off the build plate with the filament string to nozzle (especially PETG).
> 
> even with z-hop?

It is the just printed tiny part that get ripped off the build plate by the string from the nozzle. I guess it needs to cool more to get a firm grip to the build plate. Z-hop helps avoiding collisions with already printed (and cooled) parts - that is not the problem in my case.
Have tried a longer retraction setting to minimize stringing (PETG), but it makes no difference above 0.8mm.

supermerill, 2018-11-27:
> Have tried a longer retraction setting to minimize stringing (PETG), but it makes no difference above 0.8mm.

0.8mm is the nozzle size?

Area5142, 2018-11-27:
> 
> 
> > Have tried a longer retraction setting to minimize stringing (PETG), but it makes no difference above 0.8mm.
> 
> 0.8mm is the nozzle size?

No. Nozzle size is standard 0.4mm. 0.8mm is the retraction length.

> 
> > It is the just printed tiny part that get ripped off the build plate by the string from the nozzle.
> 
> Have you tried with a big first layer width (at least 150% perimeter width) ?

No - that would affect all of first layer, prevent thin walls and other mechanical details.

Just dialing down the speed to 30-50% on the printer solves the problem, but it takes ages to print as the low speed is only needed for the tiny details and end of single line. For infill, the speed can be 100%, but this way I have to dial up/down during the first layer.
When using the speed control on the printer, travel speed is also reduced, preventing the rapid move away from tiny parts and end of single lines.

ingrimsch, 2018-11-27:
Thanks for joining the topic @supermerill and @Area5142 

First layer speed in % instead of constant settings is a valid suggestion for materials like PLA, but I print PETG ~80% of the time. After ~15kg PETG my experience for successful hq prints without stringing is:

* PETG is a "leaky" filament that doesnt like big changes in extrusion rate (like slow outer perimeter speed vs. fast infill speed which is no problem when printing PLA for example). You should not vary print speed more than 20%, even better: keep a single speed for the whole print (except the first layer of course)
* Extruder needs to be well calibrated and temps need to be dialed in (too hot = worse print quality/more stringing)
* a slight underextrusion is helpful and perimeter overlap should be less than PLA
* high jerk rate and fast travel speeds help tremendously to avoid problems with leaking/stringing
* more and faster retract than PLA
* nozzle needs to be slightly higher above the bed (I usually have a Z offset of +0.04-0.05 when printing PETG) for the first layer to avoid dragging small deatils around the bed. Squishing the filament = problems with fine details (they get ripped off or dragged around)

Printing with brim helps with small screw holes or sharp corners at higher first layer speeds, but removing that in a PETG print with a lot of fine details is really awful. I try to avoid brim at all costs for that reason.

> I would like the travel speed to be reduced too as high travel speed sometime rips very small filament areas off the build plate with the filament string to nozzle (especially PETG).

My experience is the exact opposite... it needs to be fast and rapid to rip the plastic from the nozzle. :-)

> even with z-hop?

Z-hop with PETG on a cartesian printer will give you stringing and leaking. On a Delta style printer this works very well as far as I have read (no personal Delta experience here)

> When using the speed control on the printer, travel speed is also reduced, preventing the rapid move away from tiny parts and end of single lines.

Can confirm this and with PETG being that leaky it is a problem when trying to get a perfect first layer. That is one reason for actually posting this issue/feature request

> Have you tried with a big first layer width (at least 150% perimeter width) ?

That works partially actually, but doesn't help with very fine details - only lines/perimeters. Also using this approach usually ruins the perfect first layer and can create PETG specific problems (some filament collected around the nozzle which can leak on small details and rip them off of lead to stringing.

The # 1 reason for this request is PETG filament and wanting more control for a perfect first layer with that. Having more advanced control to speed and jerk settings for the first layer would be really helpful. I tried a lot of different approaches, but it always comes back to this or printing a sometimes painfully slow first layer (depending on the model). Maybe there are other filament types that would benefit from this feature. I can definately see that this is a problem that relates to more advanced users though. ;-)

Area5142, 2018-11-27:
@ingrimsch  I use almost exclusively PETG and think your problem formulation describes the problem really well.
Has also gone away from z-hop to reduce stringing.

I see the need for more fine grained first layer speed control.

What is your retract speed, I use the default 35mm/s?

ingrimsch, 2018-11-28:
Not sure if my settings are really helpful for you, as I have no genuine prusa printer, but a self built printer based on the P3steel frame running repetier firmware and a bowden extruder. For PETG I use 100mm/s retract speed with 5000 extruder acceleration, **beware:** this is a pretty high setting, not every extruder can manage a retraction that fast (hello geared extruders, I'm talking about you ;-)) and there is a good chance you will cause grinding/jams with those settings. 

I use a compact MK8 bowden since the beginning (https://www.thingiverse.com/thing:669459) driven by  a 1.8°, 1.7A Nema 17 Stepper. Never had the urge to try a direct extruder after I found my optimal settings. I have ZERO stringing in PETG and almost invisible seams/blobs on retraction spots. It is possible with PETG and bowden systems, but it was a really painful try and error for a long time to get it perfect. 

If I were you (and I had a direct extruder :-)) I'd start with 50-60 mm/s retraction speed. Bowden and leaky filaments are more complicated than direct extruders. Apart from that I use 300mm/s travel speed, advanced acceleration control (infill and default are 2000-3000, rest is 1200-1400), jerk depends on acceleration and ranges from 15.4 to 24.4. Repetier firmware dynamically adjusts jerk as high as the set acceleration requires. Control over jerk settings in the slicer depending on the print feature (perimeter, infill, etc.) would be a powerful tool to control those dynamics instead of trusting or correcting them. :-) 

**WARNING:** These are settings for a well optimized and very rigid machine. If you found this on google and try these settings on a cheap china printer you will probably get (**very**) bad results or a printer breaking into pieces/walking through the room. Just be reasonable... switching from 300-400 acceleration and jerk 3 to these settings should ring an alarm bell... ;-)

Area5142, 2018-11-29:
@ingrimsch Thank you for the specs.
I have two Prusa printers (MK2.5 + MK3 /MMU 2.0) - will try with faster retract: 50-60mm/s. Maybe that will help removing the small blobs too :)

supermerill, 2018-11-29:
You can also try 0.3mm nozzle.

From the settings point of view, duplicating every speed settings from the first layer will blob out a bit the interface. Maybe we need a system where speed can be changed inside the end/start gcode section...

ingrimsch, 2018-11-29:
@supermerill 
> duplicating every speed settings from the first layer will blob out a bit the interface

I can see that from the Interface point of view, but an additional "first layer perimeter speed" in the "Speed" --> "Modifiers" section of the interface would be the only addition needed. I see no need to duplicate the other settings. 
Or you could go the other way round... "first layer infill speed" added to modifiers, rest stays as it is now. Regarding % speeds this might even make more sense as there is only one infill type used in the first layer. Whatever makes more sense from the development point of view.

First layer print speed on any kind of printbed with any material would benefit from that, as problems like warping or little adhesion often originate from little perimeter defects or sharp turns that were printed too fast, so the filament did not get a good enough grip. This is my experience when printing with higher acceleration and jerk settings like the ones posted above.

More control over jerk settings would/should be a different issue/request supporting you're already implemented advanced acceleration control. Maybe I'll open another issue describing why a see a need for that. For now I'd be more than happy with one of the additions to the "Modifiers" section.

supermerill, 2018-12-06:
"first layer infill speed" is less confusing than perimeters, because the brim, it's a perimeter? 
The "first layer infill speed" is only for the bottom infill pattern, so it's easily understandable.

supermerill, 2018-12-10:
@ingrimsch implemented in the last release of my fork (but not very well tested)
commit for cherry-picking: ffb8cdca36f10d1bae9c9dfaaff125156da88ef2

ingrimsch, 2018-12-10:
Wow, thanks a lot for picking this up. I just downloaded the latest release of your fork and will give it a try. I think I accidentally stumbled over your thin wall pull request yesterday when researching weird slicing behavior for a part I printed some days ago (screw holes covered by a single outer perimeter had holes in them). 
Sorry, I didnt realize you were a programmer/dev before, hope you dont mind. :-)

When everything works as expected maybe we can try to get this merged into Slic3rPE? Never made a pull request before, but you seem to have (painful ;-)) experience...

supermerill, 2018-12-10:
It's up to Prusa to decide. It's their software and their time.

ingrimsch, 2018-12-26:
Hi @supermerill 
I had the time to test your implementation (and your fork) in a few prints for the last two weeks and had amazing results using faster infill speeds. When printing large or many objects at once this is a real time saver and basically "start and forget" instead of waiting for the first layer to complete (and adjust the speeds manually if needed).
Thanks again for your help, effort and invested time on this, its very much appreciated! :-)

Merry Christmas
Peter

unixunion, 2021-08-19:
Just to add, first layer travel speed limits would be great for me, I have a very fast normal movement speed of 450mm/s, and this is too fast for first layer PETG work. So I need a first layer travel override somehow .

pmarks-net, 2024-02-26:
I find it silly that there is no built-in way to reduce the first-layer travel speed.
Here is a post-processing script:

    # This PrusaSlicer post-processing script applies "First layer speed"
    # (mm/s only, % not supported) to travel moves in the first layer.
    #
    # Usage:
    #   C:\Path\To\python.exe C:\Path\To\first_layer_speed.py
    
    import os
    import re
    import sys
    
    gcode_file = sys.argv[1]
    speed = int(os.environ['SLIC3R_FIRST_LAYER_SPEED']) * 60
    
    with open(gcode_file, "r", newline="") as f:
        lines = f.readlines()
    
    move_re = re.compile(r"^(G1 X[0-9.]+ Y[0-9.]+ (?:Z[0-9.]+ )?)(F[0-9]+)(\s*)$")
    
    first_layer = False
    
    for n, line in enumerate(lines):
        if line.startswith(";LAYER_CHANGE"):
            if first_layer:
                break
            first_layer = True
    
        m = move_re.match(line)
        if m:
            prefix, old_speed, eol = m.groups()
            lines[n] = f"{prefix}F{speed} ;; was {old_speed}{eol}"
    
    with open(gcode_file, "w", newline="") as f:
        for line in lines:
            f.write(line)

TerPuc, 2025-03-06:
"First layer solid infill speed" added in PS2.9.1-alpha1. Closing.

@unixunion for "first layer travel speed" follow issue #13806

closed 2025-03-06 by TerPuc