prusa3d/PrusaSlicer #1425: [Feature Request] Join parts with "face to bed selector"

mperino, 2018-11-24:
### Version
1.41.2+Linux64

### Operating system type + version
Ubuntu kernel 4.13.0

### Behavior
This is a new feature request.  It would be very useful to have a "Join" feature, that worked similarly to the "surface to bed" selector.  You would be able to select a face, then be prompted "To bed" or "Join to Another surface".  If you selected "Another surface", it would allow you to use the same surface selector to select a surface on another object that is loaded.

bubnikv, 2019-08-01:
How is this feature supposed to work? Do you have a real world scenario?

mperino, 2019-08-01:
Yes, I do.  
The real world scenario is that sometimes model makers produce a bunch of STL that you would like to merge into one single part. The surface selector for "Place on Face" is pretty awesome, and it would be great to have a tool that is the opposite of the "Cut" that merges together 2 parts using the same selector.

So as a visual example:
2 parts on the build plate:
[image]
Select One Face:
[image]
Select the Second Face:
[image]

Then Prusa Slicer merges the two parts by aligning the two faces (face center to face center).   If there are multiple mapping orientations, it picks the one at the smallest rotation angle to merge.

closed 2026-09-29 by prusaslicer-bot[bot]

jtagcat, 2026-10-01:
unstale