prusa3d/PrusaSlicer #1426: Wipe into object only works with more than one object to print.

mavu, 2018-11-25:
### Version
1.41.2+linux64

### Operating system type + version
Debian testing

### Behavior
* _Describe the problem_
Slic3r does not generate tool change gcode when "wipe into this object" is checked in the Object settings dialog and this Object is the only one being printed.
* _Steps needed to reproduce the problem_
Add object.
Go to settings and use "load generic" to add a slab. 
Select a different extruder for the slab.
Select different extruder for the original object.
Check 'wipe into this object' on the object settings.
* _Expected Results_
Expected result would be a tool change at the apropriate moments and a "wipe wall"
* _Actual Results_
Object prints without problems but also without tool changes.

_Is this a new feature request?_
No

lukasmatena, 2018-11-25:
Setting the "wipe into this object" in the object settings dialog tells Slic3r that it can use the object for wiping - and that you therefore don't care about the color of that object. Whatever extruder you set for that object, it will be overridden whenever Slic3r decides. If there are no other objects, there is no reason for Slic3r to generate any toolchange at all. It just loads a filament and prints the whole thing.

Adding the modifier mesh does not "uncheck" the "wipe into this object" for the subpart (Slic3r does not even allow that). And therefore, whatever extruder you set.... So as far as I'm concerned, this is not a bug - the result is in agreement with the config settings.

I understand, though, that it would be more universal if Slic3r allowed to check and uncheck the "wipe into this object" setting for all the individual parts. If I recall correctly, there were some technical reasons for not allowing that - the changes to the code were quite non-trivial regardless and this would have allowed even more complete redesign of the logic. Wiping into INFILL can be set individually for the subparts.

I hope I understood, remembered and answered everything correctly.

rtyr, 2020-03-29:
No additional questions. Closing.

closed 2020-03-29 by rtyr