prusa3d/PrusaSlicer #1429: MMU2 profile hides user preset filament profiles

rncfiles, 2018-11-26:
### Version
1.41.1+Win64

### Operating system type + version
Win64

### Behavior
* _Describe the problem_
When MK3 MMU2 printer profile is selected, most user defined filament profiles are removed from the selection list. 
* _Steps needed to reproduce the problem_
With "Original Prusa i3 MK3" printer profile selected, create a new user defined filament profile.  In my case I made modifications to included filament profiles and saved them as my own.  Switch to "Original Prusa i3 MK3 MMU2" printer profile and see that under filament drop down, the user saved filament does not show.  Note that choosing "Original Prusa i3 MK3 MMU2 Single" does not have this error.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
User defined filament profiles should be available to select for MMU2 profile
* _Actual Results_
User defined filament profiles are missing
  * _Screenshots from __*Slic3r*__ preview are preferred_
MK3 profile
[image]

MMU2 profile
[image]

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

rncfiles, 2018-11-26:
Looks like this is intentional behavior.   All non MMU2 filament profiles have this line under "dependencies"
` !(printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK(2.5|3).*/ and single_extruder_multi_material)`

So any user presets derived from non MMU2 profiles will be hidden.  The three that show up for me were created in an earlier version of Slic3r so they do not have this dependency listing.

bubnikv, 2018-11-26:
This is on purpose, as the MMU2 filament profiles define a ramming
sequence, which is different from the MMU1. We only allow filament profiles
for the MMU2, which were carefully optimized for the MMU2. We are slowly
adding new filament profiles as we test these materials to be compatible
with MMU2.

If you derive your filament profile from the MMU2 compatible filament
profile, it will show up.

rncfiles, 2018-11-26:
Ok.  This makes sense.  If this was documented somewhere, I missed it.  If not, the request would be to spell this in better detail somewhere.

bubnikv, 2018-12-13:
Issue was explained, working as expected. Closing.

closed 2018-12-13 by bubnikv