prusa3d/PrusaSlicer #1430: Error with triangular infill

fromanza, 2018-11-26:
### Version
Version 1.41.1+win64

### Operating system type + version
Windows 10, latest updates

### Behavior
- Slic3r shows correct triangular infill however once the gcode executes and the printer is printing then the actual gcode paths differ.
- See photos and screenshots attached of missed infill structure as well as incorrect angles on two legs of the infill. I have only ever seen this on triangular infill.
- Printed with Prusa Original Mk3 firmware 3.4.1
- GCode viewer (http://gcode.ws/) shows correct paths however I watched the printer printing in these error regions and it does not follow these paths, it follows the as-printed paths...
- I need to try and replicate this error
- It was not a blocked nozzle
- Print completed successfully other than this weird infill error

#### STL/Config (.ZIP) where problem occurs
[28 - Drill Parts.zip](https://github.com/prusa3d/Slic3r/files/2614368/28.-.Drill.Parts.zip)

bubnikv, 2018-11-26:
It seems this is not a Slic3r issue, but a firmware issue. It looks as some
of the segments were lost by the printer firmware. Would you please try an
older version of the firmware?

Did you print from an SD card or from OctoPrint or over an USB line?

We will certainly verify it in house.

jindrichbenes, 2018-11-26:
Did you happen to have message "crash detected, continue printing" pop up and then selected "no" and started the next print?

fromanza, 2018-11-26:
@bubnikv I'll try doing this print again. If it fails then I'll slice & print the model to just include the region of interest. If it still fails then I'll flash to an older firmware & try again. Printed directly from SD & first time ever seeing anything like this. I hope to have an answer for you in 24 hours.

@jindrichbenes No messages, I watched the printer physically 'printing' and moving along those incorrect paths. When I first saw the problem I figured it was just the plastic extrudate hanging up somewhere else and snagging but in the one region shown in the attachment it just stops extruding and in the other region it literally moved along those odd angled paths.

Agree, It seems the g-code is correct but the printer has done something odd.

fromanza, 2018-11-26:
@jindrichbenes Sorry I forgot to mention, when I saw it was repeatedly doing this random infill movement,, I paused the print to take a photo then resumed the print. Resumed fine & continued to complete the print.

jindrichbenes, 2018-11-27:
This appear to be an issue fixed in the 3.5.0 fw. It som,etimes occured when window "cresh detected, continue printing" popped up and user selected "no".

[image]

fromanza, 2018-11-27:
Hmmm, I'll update FW to latest and do the print again to see if there are any differences. I don't have crash detection turned on because it doesn't work reliably on my printer.
Your prints look good but can you advise if the infill is attaching to the perimeters correctly?

My last prints (below) look better which is weird as the gcode used is exactly the same as used previously and the FW is the same. There are places where the infill doesn't attach properly and also some places that the lines are not quite straight.

[image]

[image]

[image]
[image]
[image]

bubnikv, 2019-01-29:
Did you solve your issue? Could we close the issue here?

fromanza, 2019-01-29:
Yes, please close issue. The last print seemed to print ok. Assume maybe a
bug in the printer firmware or just a bad day...

bubnikv, 2019-01-29:
Great, closing.

closed 2019-01-29 by bubnikv