prusa3d/PrusaSlicer #1431: Slic3rPE-1.41.2+linux64-full-201811221508.AppImage does not install desktop integration

kortschak, 2018-11-26:
### Version
Slic3rPE-1.41.2

### Operating system type + version
Linux Ubuntu 16.04

### Behavior
Usually, when a new version of Slic3r is first run, it will ask to update the desktop integration. This version does not do so.

macfly1202, 2018-12-16:
I close my opened issue which is the same 
https://github.com/prusa3d/Slic3r/issues/1455

rmoravcik, 2018-12-17:
It looks like a Linux Slic3rPE-1.41.2 AppImage is calling slic3r script instead of slic3r.wrapper script (that is also missing in new AppImage).

It seems that slic3r.wrapper script was doing some desktop integration magic...

Ghostbird, 2018-12-26:
This is a bit of a shame, I'd love to have this feature back in a new release, for convenience's sake.

macfly1202, 2019-02-14:
Hi, release [Slic3r PE1.41.3](https://github.com/prusa3d/Slic3r/releases/tag/version_1.41.3) still have this same issue - Prusa dev team, please correct this annoying bug.

bubnikv, 2019-02-17:
Please advice as how we can fix this annoying bug. I am out of my wits.

rmoravcik, 2019-02-17:
> Please advice as how we can fix this annoying bug. I am out of my wits.
> […](#)

The previous versions of Linux AppImage had slic3r.wrapper script, that was taking care of desktop integration and this script is missing in new Linux AppImages.

rmoravcik, 2019-02-17:
@bubnikv 

this is a content of 1.41.1 Linux AppImage, that is working fine:
```
morgan@morgan:/tmp/.mount_Slic3rGZqZER$ ls -l
celkom 38
-rwx--x--x 1 root root  1396 okt 26 14:04 AppRun
-rwx------ 1 root root 21520 okt 26 14:04 libunionpreload.so
-rw------- 1 root root   174 okt 26 14:04 slic3r.desktop
-rw------- 1 root root 16469 okt 26 14:04 slic3r.png
drwxr-xr-x 6 root root     0 okt 26 14:04 usr
morgan@morgan:/tmp/.mount_Slic3rGZqZER$ cat slic3r.desktop 
[Desktop Entry]
Name=Slic3r
Exec=slic3r.wrapper --gui
Icon=slic3r
Terminal=false
Type=Application
Categories=Graphics;3DGraphics;
X-AppImage-Version=1.41.1+linux64.glibc2.14
morgan@morgan:/tmp/.mount_Slic3rGZqZER$ ls -l usr/bin/
celkom 40
drwx------ 2 root root     0 okt 26 14:04 bin
drwx------ 3 root root     0 okt 26 14:04 lib
drwx------ 7 root root     0 okt 26 14:04 resources
-rwx------ 1 root root   148 okt 26 14:04 slic3r
-rwx------ 1 root root 30662 okt 26 14:04 slic3r.pl
-rwx--x--x 1 root root 10259 okt 26 14:04 slic3r.wrapper
morgan@morgan:/tmp/.mount_Slic3rGZqZER$ 
```
and this is 1.41.2, that is missing slic3r.wapper script:
```
morgan@morgan:/tmp/.mount_Slic3rQiyRCA$ ls -l
celkom 38
-rwx--x--x 1 root root  1396 nov 22 16:22 AppRun
-rwx------ 1 root root 21520 nov 22 16:23 libunionpreload.so
-rw------- 1 root root   166 nov 22 16:23 slic3r.desktop
-rw------- 1 root root 16469 nov 22 16:22 slic3r.png
drwxr-xr-x 6 root root     0 nov 22 16:22 usr
morgan@morgan:/tmp/.mount_Slic3rQiyRCA$ cat slic3r.desktop 
[Desktop Entry]
Name=Slic3r
Exec=slic3r --gui
Icon=slic3r
Terminal=false
Type=Application
Categories=Graphics;3DGraphics;
X-AppImage-Version=1.41.2+linux64.glibc2.14
morgan@morgan:/tmp/.mount_Slic3rQiyRCA$ ls -l usr/bin/
celkom 30
drwx------ 2 root root     0 nov 22 16:23 bin
drwx------ 3 root root     0 nov 22 16:22 lib
drwx------ 7 root root     0 nov 22 16:22 resources
-rwx------ 1 root root   148 nov 22 16:22 slic3r
-rwx------ 1 root root 30662 nov 22 16:22 slic3r.pl
morgan@morgan:/tmp/.mount_Slic3rQiyRCA$ 
```

ardenpm, 2019-05-27:
This issue still exists in the PrusaSlicer 2.0.0 release.

bubnikv, 2019-06-03:
@probonopd please help again.

Reading the last comment of
https://discourse.appimage.org/t/standardise-removal-of-appimage/62/9

it seems that the appimage integration is no more done by the appimage itself, but by appimaged daemon (or the AppImageLauncher tool), which the user has to install on his computer, right?

ardenpm, 2019-06-03:
In my case I had attempted installing the appimaged package (I’m on Ubuntu 16.04) and following various guides placed the AppImage in its various watch directories but it did not pick it up.

probonopd, 2019-06-03:
It seems like `usr/share/applications/prusaslicer.desktop` is missing; hence `appimaged` cannot integrate it into the system. @bubnikv where can I see the script that generates the AppImage?

bubnikv, 2019-06-03:
This is the content of the PrusaSlicer.yml
# This is an AppImage recepie by @probonopd 
# https://github.com/AppImage/AppImages/blob/master/YML.md

app: PrusaSlicer
union: true

ingredients:
  dist: oldstable
  sources:
    - deb http://ftp.de.debian.org/debian/ oldstable main
  packages: freeglut3
  script:
    - echo "${SLIC3R_BUILD_ID}" | sed -e 's|-|.|g' > VERSION
    - echo "Version"
    - cat VERSION
    - echo "multiarch-support" >> excludedeblist

script:
  - pwd
  - mkdir -p usr/bin
  - cp -R ${DIST_DIR_FULL}/* ./usr/bin
  - cp ./usr/bin/resources/icons/Slic3r.png PrusaSlicer.png
  - mv ./usr/lib/x86_64-linux-gnu/lib* ./usr/bin/bin/ # Their wrapper loads libraries only from there
  - cp ${BUILD_UTILS_DIR}/PrusaSlicer.desktop ./
  - cp ${BUILD_UTILS_DIR}/PrusaSlicer.AppRun ./AppRun

The PrusaSlicer.desktop lands to the root of the AppImage
[image]

probonopd, 2019-06-03:
OK, please add to the end:

```
- mkdir -p ./usr/share/applications && cp ${BUILD_UTILS_DIR}/PrusaSlicer.desktop ./usr/share/applications
```

That should do the trick.

probonopd, 2019-06-03:
By the way, is `union: true` really still needed? I consider it to be a bit of a hack for applications where the authors don't want to make changes; in your case you should be able to do without it because you can make the application look up its resources in paths relative to the main binary's directory (if it doesn't already do that).

azubieta, 2019-06-04:
> It seems like `usr/share/applications/prusaslicer.desktop` is missing; hence `appimaged` cannot integrate it into the system. @bubnikv where can I see the script that generates the AppImage?

This doesn't seems to be the issue here. `appimaged` performs the system integration using `libappimage` and this support desktop files in the root of the AppDir. I guess that we would need the `appimaged` logs and the generated desktop file in the user home (if any) to properly check. Also would be nice to know the system in which this issue is happening.

By the way feel free to ping me on the #appimage channel on freenode for a closer review of this issue.

probonopd, 2019-06-04:
Sorry, I stand corrected. What is the best way for a user who experiences this issue to help debugging @azubieta?

probonopd, 2019-06-16:
```
me@host:~/Desktop$ /isodevice/Applications/PrusaSlicer-2.0.0-rc2+linux64-201905190649.AppImage --appimage-extract '*.png'
squashfs-root/PrusaSlicer.png
me@host:~/Desktop$ file squashfs-root/PrusaSlicer.png 
squashfs-root/PrusaSlicer.png: PNG image data, 154 x 154, 8-bit/color RGBA, non-interlaced
```

154x154 is not a valid icon size and hence the icon does not show up. Please use, e.g., 256x256 or 128x128.

bubnikv, 2019-06-17:
Great, thanks a lot for investigation.

bubnikv, 2019-08-19:
@probonopd I have updated the build scripts to use the 128x128 picture for desktop integration. Let's keep our fingers crossed.

probonopd, 2019-08-19:
Thanks @bubnikv

bubnikv, 2019-08-23:
Please verify with the upcoming PrusaSlicer 2.1.0-beta.

kortschak, 2019-12-23:
Still not working with 2.2.0-alpha2.

bubnikv, 2020-03-20:
The PrusaSlicer.png in the root of the AppImage is
128x128 8 bit color RGBA, non interlaced

probonopd, 2020-03-20:
I cannot reproduce this issue using https://github.com/probonopd/go-appimage/tree/master/src/appimaged.

[image]

kortschak, 2020-03-20:
Maybe this is a documentation issue then. Prusa should provide information for the correct way to install the software.

bubnikv, 2020-03-21:
>  Maybe this is a documentation issue then. Prusa should provide
information for the correct way to install the software.

Maybe it is your distro issue? Maybe your appimaged is misconfigured on
your box?

kortschak, 2020-03-21:
>Maybe it is your distro issue? Maybe your appimaged is misconfigured on
your box?

This is certainly something that would be helped with adequate documentation for installation, no?

kortschak, 2020-03-21:
To be completely clear, I can find no obvious documentation for desktop integration for the Prusa Slicer appImage release, and no indication (prior to the comments above) that appimaged should be used.

As an aside, I have to say that [this](https://github.com/AppImage/appimaged#monitored-directories) is terrifying.

probonopd, 2020-03-21:
> Prusa should provide information for the correct way to install the software.

AppImages do not need to be "installed". Just download, make executable, run.

If you want desktop integration (icons, menu entries) you need either a Linux distribution that integrates AppImages with the system out of the box, or an optional daemon/system that integrates AppImages (not just PrusaSlicer but all of them) into the system, e.g., appimaged or AppImageLauncher. This has nothing to do with Prusa or Prusa documentation. If anything, you can blame AppImage documentation or your distribution's desktop environment for not doing it natively.

kortschak, 2020-03-21:
Thanks. I understand how AppImage applications work, and how desktop integration works. The title of the issue remains accurate, and could be at least partially mitigated by adding documentation (or links to documentation) explaining (as you have done above) how to perform desktop integration since that behaviour has been removed from the applmage applications' first run.

I agree, the defect does largely exist in AppImage's documentation. While the existence of appimaged and AppImageLauncher is described in the documentation, it could be made more clearly visible. A link in the Prusa Slicer install (for want of a better word) instructions would help here. Having said that, the approach appimaged takes is truly horrifying, and it's sad to see a nice simple approach of AppImage now picking up additional weight, requiring these extra toolings.

probonopd, 2020-03-21:
@kortschak thanks for your response. Please keep in mind that desktop integration is still entirely optional. You can still just double-click AppImages from the file manager. We (the AppImage team) are always happy to think about how to improve. Maybe we should continue over at https://github.com/AppImage/AppImageKit/issues.

spuder, 2020-12-13:
As a newish linux desktop user I found this really frustrating that prusa slicer wouldn't show up in the launcher. For anyone else looking for a quick workaround, the snap store provides an install of prusa slicer that includes the desktop file 

https://snapcraft.io/prusa-slicer

```
sudo snap install prusa-slicer
```

macfly1202, 2020-12-13:
amazing and it's works ! so cool. Thanks for sharing @spuder

probonopd, 2020-12-13:
> As a newish linux desktop user I found this really frustrating that prusa slicer wouldn't show up in the launcher. 

This is because most desktop environments do not offer native AppImage integration yet. But optional tools like AppImageLauncher and appimaged can integrate AppImages into the menu automatically for you.

ardenpm, 2020-12-13:
The AppImage ecosystem is a bit frustrating at the moment. To get the integration as mentioned earlier you really need appimaged now, however the official repo for that says it is deprecated and they direct you to [this one](https://github.com/probonopd/go-appimage) which is clearly experimental and pre-release.

In any case I did download the appimaged from the new repo and ran it (just like another AppImage) which installs the daemon. After that it adds the stuff automatically when you drop an AppImage in a watched folder (Downloads). That is what got me at first since I was putting them in a specific place that wasn't being watched.

Really though, the Appimage project hasn't done a good job of keeping the desktop integration stuff simple. You now need to wade through multiple Github repos to work it out.