prusa3d/PrusaSlicer #1433: Fan behavior false

hevilp, 2018-11-26:
### Version
1.41.2
### Operating system type + version
Win64

### Behavior
The Fan should go on second layer.

In the gcode the first M106 command is near 1.8 mm, which is the 6. layer.
[Gringerbread_man_cokie_cutter.gcode.txt](https://github.com/prusa3d/Slic3r/files/2616951/Gringerbread_man_cokie_cutter.gcode.txt)
[image]

supermerill, 2018-11-27:
Could not reproduce.

Can you send the .3mf?  
when your reproduce the bug go to plater -> export as 3mf, and link it here.  
Thank you.

supermerill, 2018-11-27:
I put 1 at "disable fan for the first .. layers", export a gcode on a random object i have and seen that the M106 is before the first layer.

hevilp, 2018-11-27:
Not my real settings at home, very very late M106 fan appearance in this file too
[Gingerbread__Cookie_Cutter.3mf.txt](https://github.com/prusa3d/Slic3r/files/2620414/Gingerbread__Cookie_Cutter.3mf.txt)

hevilp, 2018-11-27:
I tested it more,

I have to set "Enable fan if layer print time is below" to very high value.

The sense of this option is, that the filament cools down to normal temperature, without an active fan, because the layer printing time is long enough, right?
if my print time is above 60 seconds (default value), the fan turns off?
If I set it to 300, the behavior should be right.

supermerill, 2018-11-27:
You forgot to check "fan always on"

closed 2018-11-28 by hevilp