prusa3d/PrusaSlicer #1436: Slic3r crashed without error

zonk0101, 2018-11-28:
### Version
_Version of Slic3r used goes here_
Slic3rPE-1.41.1+win64-full-201810261148
Slic3rPE-1.41.2+win64-full-201811221508
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
OS Name	Microsoft Windows 10 Pro
Version	10.0.17134 Build 17134

### Behavior
* _Describe the problem_
Slic3er closes itself after clicking on "Slice now"
Config doesn't seem to have an impact.
I have modified the STL and with a minor tweak it is slicing fine:
- Object wall-thickness in openscad (not config) of 0.6 crashes, but 0.59 is working fine.

* _Steps needed to reproduce the problem_
Import attached STL and click on Slice now.
[Files.zip](https://github.com/prusa3d/Slic3r/files/2625130/Files.zip)

* _Expected Results_
Sliced object :)
* _Actual Results_
Slic3erPE closed without any warning or error

_Is this a new feature request?_
no

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[Files.zip](https://github.com/prusa3d/Slic3r/files/2625130/Files.zip)

FidelCapo, 2018-12-04:
I found that if I set the skirt height to 1 layer and the Elephant foot compensation to 0.3mm, the slic3r with the crashes.stl object will crash. When I change the skirt height to 3 in your config slic3r will not crash.

Tested with 1.41.2

FidelCapo, 2018-12-14:
@bubnikv yes. I can reproduce crash with 1.42.0-alpha

FidelCapo, 2019-08-26:
I cannot reproduce it with PrusaSlicer 2.1.0-beta2. I think this issue is fixed now.
Closing.

closed 2019-08-26 by FidelCapo